microsoft/HoloLens2ForCV #1: Detection works but are the virtual content (coordinate system and ray) offset with some transformation

qian256, 2020-08-28:
Awesome!
When I run the CameraWithCVAndCalibration sample, I can see the two coordinate systems and two rays showing the direction of marker from the coordinate systems of the two VLC cameras. Is it an issue or a feature that the rays do not align with the actual physical device? It looks like the origin of the whole virtual scene is offseted with a transformation.

dorinung, 2020-08-28:
Thank you for your feedback. This is by design. The app shows a "birds eye view" of the ray triangulation. I am working on additions to the app that will render in the device coordinate system.

qian256, 2020-08-28:
Thank you. The additional feature to superimpose on the physical world (device coordinate system) would be very helpful.

closed 2020-08-28 by qian256