microsoft/HoloLens2ForCV #4: Can't download StreamRecorder data

murrdpirate, 2020-08-28:
Thanks for posting the repo!

I'm able to run the StreamRecorder and I can see `Depth Long Throw.tar` (among other files) in the Device Portal, but when I try to save them to my PC, the browser times out. 

Perhaps similarly, when I run `recorder_console.py`, I get the following error: `urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)>`

I did install the rootcertificate from the HoloLens2 onto my PC, if that's relevant.

MeganSage, 2020-08-30:
Add this to your code:
**import ssl
ssl._create_default_https_context = ssl._create_unverified_context**
And  uncheck **SSL connection** from **Preferences** under **system** at **Windows Device Portal**.
Thats my solution!

kysucix, 2020-08-31:
hi @murrdpirate I wasn't able to reproduce your issue. Please try the fix proposed by megansage and let me know if that's fixing your problem.

murrdpirate, 2020-08-31:
Unchecking the **SSL connection** as @MeganSage suggested made it so I can access the recordings via `recorder_console.py`. And this actually worked immediately - before I attempted to add the 2 lines of code. I still cannot download via the Device Portal, but this is enough for me. Thank you!

closed 2020-08-31 by murrdpirate

ylsislove, 2021-03-04:
I solved the SSL error according to the above method, but I still cannot download the StreamRecorder data via `recorder_console.py`. I got the following error: `urllib.error.HTTPError: HTTP Error 401: Unauthorized`. I checked my username and password to make sure it was correct when I ran `recorder_console.py`, but I still got `Error 401: Unauthorized`.

Any idea what is causing the issue? Any help is appreaciated!

ylsislove, 2021-03-04:
I solved it. I noticed that the IP address for the HoloLens Device Portal in `recorder_console.py` is `127.0.0.1:10080`, but I entered `http://127.0.0.1:10080` in the browser, it will be redirected to `https://127.0.0.1:10443/`. So I changed the `self.url` in the script to `https://127.0.0.1:10443/`, and it worked. Now I can download the StreamRecorder data from Hololens! perfect!

My problem may be caused by using USB connection to download data. Maybe connecting Hololens Device Portal with WiFi will not cause my problem.