microsoft/HoloLens2ForCV #6: Unable to run Sensor Visualization sample

MediosZ, 2020-08-29:
I tried to run Sensor Visualization sample but failed with the Exception: WinRT originate error - 0x80004002 : 'No such interface supported'.

Windows SDK version: 10.0.19041.0
Hololens OS build: 10.0.19041.1113

murrdpirate, 2020-08-29:
I think you need to update your HoloLens OS build to 19041.1364 (see slide 6 here:  https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ECCV2020-Tutorial/ECCV2020-ResearchMode-Api.pdf)

I was able to get this update with the developer preview beta, which you can enable in the settings.

MediosZ, 2020-08-29:
Thanks! 
I updated the OS without the developer preview beta, maybe that's the reason I didn't get the update you mentioned.
I'll try that.

MediosZ, 2020-08-29:
After upgrading to 10.0.19041.1369, I cannot run the sample as well.
This time, the exception is Exception thrown at 0x00007FFF732E39EC (KernelBase.dll) in SensorVisualization.exe: WinRT originate error - 0x8000FFFF : 'Catastrophic failure'.

murrdpirate, 2020-08-29:
Hmm, I updated to exactly the 19041.1364 version mentioned in the slides. There are two options for the developer preview, something like alpha and beta. Maybe you selected alpha?

MediosZ, 2020-08-29:
I selected the dev channel not the beta channel, which means I can get less reliable builds.
I'll try to switch beta channel to see if i can get the same build with you.

dorinung, 2020-08-29:
Please make sure Research mode is enabled in the device portal. See the Setup section in https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ECCV2020-Tutorial/ECCV2020-ResearchMode-Api.pdf (slides 7 and 8)

MediosZ, 2020-08-30:
Sorry I forget to enable the Research mode after upgrading the OS and now it's working.

Here are several tips for those who are new to hololens:
1. You should join the Windows Insider Program to get preview builds.
2. After that you can see the Research mode option in Device Portal. 
3. Research mode is different from Recording mode. You can only see Recording mode when you are not with preview builds.
Hope this can help someone who cannot enable the Research mode.

Thanks! @dorinung @murrdpirate 

I also want to know if we can use Research mode APIs using C# or it is only for c++ ?

closed 2020-08-30 by MediosZ