microsoft/HoloLens2ForCV #7: Live stream to Desktop for online processing?

agroenenberg, 2020-09-01:
Hi, 

Is it possible to live stream sensor streams to my desktop for online processing? I'm trying out some registration algorithms but I'm not sure whether the HPU can handle them? Thanks! (And thank you for this Git, can't wait to get everything up and running)

kysucix, 2020-09-03:
Hi agroenenberg,

streaming to desktop is something we're planning to release soon.

We need time to cleanup code and test.

Silvano

rooi, 2020-09-03:
Hi Silvano,

I just noticed this issue and you reply and I'm very interessted in this functionality as well.
Could you, perhaps, give in indication on what you mean by soon? Days, weeks, months? Not to pin you on this of course, but just to take this into account.

Thanks!

Roy

kysucix, 2020-09-03:
hi Roy,

We can't promise a release date but it could be within a month, even less, the code is there, but we need a bit to clean it up 
first :)

rooi, 2020-09-04:
Thanks Silvano!

rooi, 2020-09-30:
Hi Silvano,

How is het going with the cleanup going? Obviously I'm meaning to ask you when we can expect the streaming functionality ;)
You mentioned the code you plan to use is already there, can you give an indication which code/methods/examples you plan to use (the one that's in HololensForCV for HL1?). Then we can perpare are own software for the stream.

Thanks!

Roy

Lancename, 2020-10-15:
Hi Silvano,
I wonder if  the streaming app and the remote app  are able to work at the same time ?    It is very important for me.  
Thanks!  
Lance

agroenenberg, 2020-10-29:
Hi @kysucix , do you maybe have any updates on this? Thanks in advance!

dennisritter, 2020-11-19:
Hi Silvano,

are there any updates on this? A Streaming-to-Desktop solution would be really helpful for my current work.
Thanks in advance!

Dennis

agroenenberg, 2020-11-23:
https://github.com/petergu684/HoloLens2-ResearchMode-Unity/ Check out this github

cgsaxner, 2020-12-19:
Hi,
I started to implement my own Streamer of RM sensor data and video camera frames to desktop from a Unity Plugin. Currently, it works for AHAT and the photo video camera:
https://github.com/cgsaxner/HoloLens2-Unity-ResearchModeStreamer
It is still work in progress, but maybe a place to start for others interested in this functionality.

zovathio, 2021-02-19:
Hi @kysucix, thanks for your work! Do you maybe have any updates on this? Thanks in advance!

sitzikbs, 2021-03-04:
Hi @kysucix, voting up the usefulness of a stream research mode streams to desktop.

dogadogan, 2021-11-29:
Do you have any updates on this? Thank you!