microsoft/HoloLens2ForCV #8: StreamRecorder Error - "Unable to activate Windows Store app"

nickgreene, 2020-09-01:
Hello and thanks very much for this repository.

I am able to run the CalibrationVisualization sample, however I am running into errors when I try to run the StreamRecorder sample (and similar errors for the other samples). 

When I press "Start with debugging" Visual Studio 2019 builds and deploys without error. During runtime, sometimes the following message is displayed in an error window:

```
Unable to activate Windows Store app
'31fa9d1a-a222-4878-a6fc-77aff92195b5_ph1m9x8skttmg!App'. The
activation request failed with error 'Invalid value for registry'.

See help for advice on troubleshooting the issue
```

Othertimes, it reaches a breakpoint in thread:
[image]

<details>

<summary>Here is the full debug output from StreamRecorder</summary>

```
'SensorVisualization.exe' (Win32): Loaded 'U:\USERS\DefaultAccount\AppData\Local\DevelopmentFiles\C33DB2CF-841E-466C-8CC0-FB956BC75B07VS.Debug_ARM64.Nick\SensorVisualization.exe'. Symbols loaded.
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\ntdll.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\KernelBase.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\combase.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\ucrtbase.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\rpcrt4.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\kernel32legacy.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\oleaut32.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\msvcp_win.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\d2d1.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\msvcrt.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\d3d11.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\win32u.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\dxgi.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\DWrite.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\forwarders\kernel32.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'P:\WindowsApps\Microsoft.VCLibs.140.00.Debug_14.0.27810.0_arm64__8wekyb3d8bbwe\msvcp140d_app.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'P:\WindowsApps\Microsoft.VCLibs.140.00.Debug_14.0.27810.0_arm64__8wekyb3d8bbwe\vcruntime140d_app.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'U:\USERS\DefaultAccount\AppData\Local\DevelopmentFiles\C33DB2CF-841E-466C-8CC0-FB956BC75B07VS.Debug_ARM64.Nick\ucrtbased.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\gdi32.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\gdi32min.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\MinUser.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\msvcp110_win.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\SHCore.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\bcryptprimitives.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\procthreadexthost.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\kernel.appcore.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\twinapi.appcore.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\sechost.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\WinTypes.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\dcomp.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\CoreUIComponents.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\ntmarta.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\advapi32legacy.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\CoreMessaging.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\bcrypt.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\ws2_32.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\Windows.StateRepositoryCore.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\Windows.UI.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\TextInputFramework.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\WindowManagementAPI.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\InputHost.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\propsys.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\OneCoreUAPCommonProxyStub.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\d2d1debug3.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\WindowsCodecs.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\Windows.Gaming.Input.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\cfgmgr32.dll'. 
[... 55 lines cut ...]
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\crypt32.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\ksuser.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\RTWorkQ.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\umpdc.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\Windows.Web.dll'. 
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\iertutil.dll'. 
Exception thrown at 0x00007FFC0B5439EC (KernelBase.dll) in SensorVisualization.exe: WinRT originate error - 0x80004002 : 'No such interface supported'.
Exception thrown at 0x00007FFC0B5439EC in SensorVisualization.exe: Microsoft C++ exception: winrt::hresult_no_interface at memory location 0x0000004DBB0FD650.
Exception thrown at 0x00007FFC0B5439EC in SensorVisualization.exe: Microsoft C++ exception: [rethrow] at memory location 0x0000000000000000.
onecoreuap\shell\coreapplication\application\lib\coreapplicationview.cpp(371)\twinapi.appcore.dll!00007FFC0640E598: (caller: 00007FFC0BB41064) ReturnHr(1) tid(1804) 80004002 No such interface supported
    CallContext:[\AppRunOrActivateView] 
onecoreuap\shell\coreapplication\application\lib\coreapplicationviewagilecontainer.cpp(474)\twinapi.appcore.dll!00007FFC063FF4C0: (caller: 00007FFC06386500) ReturnHr(2) tid(1924) 80004002 No such interface supported
    CallContext:[\AppRunOrActivateView] 
onecoreuap\shell\coreapplication\application\lib\coreapplicationviewagilecontainer.cpp(347)\twinapi.appcore.dll!00007FFC063FC964: (caller: 00007FFC063870B4) ReturnHr(3) tid(1924) 80004002 No such interface supported
    CallContext:[\AppRunOrActivateView] 
Exception thrown at 0x00007FFC0B5439EC in SensorVisualization.exe: Microsoft C++ exception: Cn::XH at memory location 0x0000004DBB2FF200.
Exception thrown at 0x00007FFC0B5439EC (KernelBase.dll) in SensorVisualization.exe: WinRT originate error - 0x8000000B : 'The operation attempted to access data outside the valid range'.
'SensorVisualization.exe' (Win32): Loaded 'C:\Windows\system32\threadpoolwinrt.dll'. 
```

</details>

My development machine is Windows 10 Version 1909 (Build 18363.1016). The Hololens 2 is build 10.0.19041.1113

Any Ideas for how I should proceed?

dorinung, 2020-09-01:
For research mode OS build 19041.1364 or later is needed on Hololens. Research mode also has to be enabled in the device portal. Please see slides 6, 7 and 8 here:
https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ECCV2020-Tutorial/ECCV2020-ResearchMode-Api.pdf

agroenenberg, 2020-09-02:
You need to sign up w/ the Windows Insider Program, select beta developer mode, then install the new OS. After that, in the device portal, enable research mode (This is different than recording mode!).

closed 2020-09-02 by nickgreene

kjayarajah, 2022-05-08:
I'm going through this same issue even after updating the OS to the beta preview and enabling both the research and recording modes. Any help would be greatly appreciated, stuck with this for days!

agroenenberg, 2022-05-08:
Did you disable ssh connection in the portal?