microsoft/HoloLens2ForCV #10: Trouble getting AHAT Depth Frame in SensorVisualization

aal-rvr, 2020-09-04:
In SensorVisualizationScenario.cpp, removing the line: `#define DEPTH_USE_LONG_THROW` should make the sample work for DEPTH_AHAT.
When I do that, the line: hr = `pSensorFrame->QueryInterface(IID_PPV_ARGS(&pDepthFrame));` in SlateCameraRenderer.cpp returns depthFrame as NULL. Am I missing something?

In `SensorVisualizationScenario::~SensorVisualizationScenario()`, m_pAHATSensor is never Released. I believe this is a typo as m_pLTSensor is repeated.

Thanks.

dorinung, 2020-09-04:
There is a bug where the depth camera stops producing AHAT frames after the device goes into standby following use of the depth camera. If you reboot the device and run the app again do you see AHAT frames? We are working on a fix for this issue.

aal-rvr, 2020-09-04:
It does work on the reboot. After rebooting after adding `m_pAHATSensor->Release()` in the SensorVisualizationScenario destructor I get the depth frames even without the reboot.

There is however a new issue, after a few seconds into running the program with AHAT depth, it crashes with the following error:

`Exception thrown at 0x00007FFC51FF39EC (KernelBase.dll) in SensorVisualization.exe: WinRT originate error - 0x8007050B : 'The process hosting the driver for this device has been terminated.'.`

dorinung, 2020-09-04:
I have not seen this crash yet. I will try a repro. Would you be able to copy a stack at the crash? Thanks.

aal-rvr, 2020-09-04:
This is all that shows in the Call Stack `KernelBase.dll!00007ffc520952a0()` It also says that KernelBase.pdb is not loaded.
On Continuing, It gives a similar error for ResearchModeApi.dll with a similar Call Stack.

That's the Exception its throwing:
`Unhandled exception at 0x00007FFC38A9A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.`

Nothing shows up in the Call Stack, but here's the Debug Output:
```
onecoreuap\analog\input\spatialobjectddiclient\dll\perceptiondevicehandle.cpp(707)\PerceptionDevice.dll!00007FFC4FA84158: (caller: 00007FFC4FA82968) ReturnHr(1) tid(908) 8007050B The process hosting the driver for this device has been terminated.
onecoreuap\analog\input\spatialobjectddiclient\dll\perceptiondevicehandle.cpp(40)\PerceptionDevice.dll!00007FFC4FA829A8: (caller: 00007FFC4FA7F794) ReturnHr(2) tid(908) 8007050B The process hosting the driver for this device has been terminated.
onecoreuap\analog\input\holographicdriverclientlib\lib\spatialgraphdriverclient.cpp(2114)\Windows.Mirage.dll!00007FFC488FBBF4: (caller: 00007FFC488FBCA4) ReturnHr(1) tid(1b88) 80070006 The handle is invalid.
    CallContext:[\ClientComponentManager_OnSensorInterestCallback\DestroySpatialGraphNodeReference] 
onecoreuap\analog\input\holographicdriverclientlib\lib\spatialgraphdriverclient.cpp(2141)\Windows.Mirage.dll!00007FFC488FBCC4: (caller: 00007FFC488FB2E0) ReturnHr(2) tid(1b88) 80070006 The handle is invalid.
    CallContext:[\ClientComponentManager_OnSensorInterestCallback\DestroySpatialGraphNodeReference] 
onecoreuap\analog\input\spatialobjectddiclient\dll\perceptiondevicefactory.cpp(402)\PerceptionDevice.dll!00007FFC4FA7F7D8: (caller: 00007FFC48756150) ReturnHr(3) tid(908) 8007050B The process hosting the driver for this device has been terminated.
analog\input\mirage\publicapi\spatialinteractionsourcespatialobject.cpp(102)\Windows.Mirage.dll!00007FFC48812E44: (caller: 00007FFC48812A4C) ReturnHr(3) tid(908) 8007050B The process hosting the driver for this device has been terminated.
SensorVisualization.exe has triggered a breakpoint.

Exception thrown at 0x00007FFC38A9A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.

The thread 0x908 has exited with code 0 (0x0).
Unhandled exception at 0x00007FFC38A9A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.

Exception thrown at 0x00007FFC38A9A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.

Unhandled exception at 0x00007FFC38A9A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.

Exception thrown at 0x00007FFC38A9A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.
```
Hope that helps.

aal-rvr, 2020-09-05:
> This is all that shows in the Call Stack `KernelBase.dll!00007ffc520952a0()` It also says that KernelBase.pdb is not loaded.
> On Continuing, It gives a similar error for ResearchModeApi.dll with a similar Call Stack.
> 
> That's the Exception its throwing:
> `Unhandled exception at 0x00007FFC38A9A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.`
> 
> Only the dll shows up in the Call Stack, but here's the Debug Output:
> 
> ```
> onecoreuap\analog\input\spatialobjectddiclient\dll\perceptiondevicehandle.cpp(707)\PerceptionDevice.dll!00007FFC4FA84158: (caller: 00007FFC4FA82968) ReturnHr(1) tid(908) 8007050B The process hosting the driver for this device has been terminated.
> onecoreuap\analog\input\spatialobjectddiclient\dll\perceptiondevicehandle.cpp(40)\PerceptionDevice.dll!00007FFC4FA829A8: (caller: 00007FFC4FA7F794) ReturnHr(2) tid(908) 8007050B The process hosting the driver for this device has been terminated.
> onecoreuap\analog\input\holographicdriverclientlib\lib\spatialgraphdriverclient.cpp(2114)\Windows.Mirage.dll!00007FFC488FBBF4: (caller: 00007FFC488FBCA4) ReturnHr(1) tid(1b88) 80070006 The handle is invalid.
>     CallContext:[\ClientComponentManager_OnSensorInterestCallback\DestroySpatialGraphNodeReference] 
> onecoreuap\analog\input\holographicdriverclientlib\lib\spatialgraphdriverclient.cpp(2141)\Windows.Mirage.dll!00007FFC488FBCC4: (caller: 00007FFC488FB2E0) ReturnHr(2) tid(1b88) 80070006 The handle is invalid.
>     CallContext:[\ClientComponentManager_OnSensorInterestCallback\DestroySpatialGraphNodeReference] 
> onecoreuap\analog\input\spatialobjectddiclient\dll\perceptiondevicefactory.cpp(402)\PerceptionDevice.dll!00007FFC4FA7F7D8: (caller: 00007FFC48756150) ReturnHr(3) tid(908) 8007050B The process hosting the driver for this device has been terminated.
> analog\input\mirage\publicapi\spatialinteractionsourcespatialobject.cpp(102)\Windows.Mirage.dll!00007FFC48812E44: (caller: 00007FFC48812A4C) ReturnHr(3) tid(908) 8007050B The process hosting the driver for this device has been terminated.
> SensorVisualization.exe has triggered a breakpoint.
> 
> Exception thrown at 0x00007FFC38A9A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.
> 
> The thread 0x908 has exited with code 0 (0x0).
> Unhandled exception at 0x00007FFC38A9A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.
> 
> Exception thrown at 0x00007FFC38A9A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.
> 
> Unhandled exception at 0x00007FFC38A9A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.
> 
> Exception thrown at 0x00007FFC38A9A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.
> ```
> 
> Hope that helps.

I ran the program multiple times and every time I ran it with AHAT, it crashed with the above exceptions.
After my most recent run, depth frames stopped arriving (probably caused by me playing around with the AHAT and LT depth) which started working again after a reboot. Up until then it gave me depth frames every time without needing a reboot. Thanks.

dorinung, 2020-09-05:
Thanks for the diagnostics information. What build are you on? I would like to try the same build.

aal-rvr, 2020-09-05:
OS Build: 10.0.19041.1373

qian256, 2020-09-05:
I have encountered same above issue. Same OS build. 10.0.19041.1373

dorinung, 2020-09-10:
I was not able to repro the crash on 19041.1377.

aal-rvr, 2020-09-12:
After updating to 19041.1377 the program runs for a slightly longer time without crashing, but crashes with the similar exceptions thrown.

```
Exception thrown at 0x00007FF93B1339EC (KernelBase.dll) in SensorVisualization.exe: WinRT originate error - 0x8007050B : 'The process hosting the driver for this device has been terminated.'.
onecoreuap\analog\input\holographicdriverclientlib\lib\spatialgraphdriverclient.cpp(2114)\Windows.Mirage.dll!00007FF93157BBF4: (caller: 00007FF93157BCA4) ReturnHr(1) tid(15c8) 80070006 The handle is invalid.
    CallContext:[\ClientComponentManager_OnSensorInterestCallback\DestroySpatialGraphNodeReference] 
onecoreuap\analog\input\holographicdriverclientlib\lib\spatialgraphdriverclient.cpp(2141)\Windows.Mirage.dll!00007FF93157BCC4: (caller: 00007FF93157B2E0) ReturnHr(2) tid(15c8) 80070006 The handle is invalid.
    CallContext:[\ClientComponentManager_OnSensorInterestCallback\DestroySpatialGraphNodeReference] 
SensorVisualization.exe has triggered a breakpoint.

Exception thrown at 0x00007FF93B1339EC (KernelBase.dll) in SensorVisualization.exe: WinRT originate error - 0x80070006 : 'The handle is invalid.'.
SensorVisualization.exe has triggered a breakpoint.

Exception thrown at 0x00007FF93B1339EC in SensorVisualization.exe: Microsoft C++ exception: winrt::hresult_error at memory location 0x000000639B3FFAE0.

Exception thrown at 0x00007FF91888A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.

The thread 0x4a8 has exited with code 0 (0x0).
onecoreuap\analog\input\holographicdriverclientlib\lib\spatialgraphdriverclient.cpp(254)\Windows.Mirage.dll!00007FF931579800: (caller: 00007FF9315442F4) ReturnHr(3) tid(15c8) 80070057 The parameter is incorrect.
Unhandled exception at 0x00007FF91888A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.
```

Apart from that, he other bug where `SlateCameraRenderer::UpdateTextureFromCameraFrame()` stops getting depth frame still happens once in a while, and gets fixed on rebooting the device

aal-rvr, 2020-09-16:
OS Build: 19041.1377
Firmware: 11971.01.01.200609F

I did a clean checkout of the repository.
Only commented out the line `#define DEPTH_USE_LONG_THROW`, so I could get AHAT
Rebooted the device, uninstalled and reinstalled the App.

The program ran for about 7 mins 30 sec. This time it showed me where it broke:
In base.h [Line:4871] `throw hresult_error(result, take_ownership_from_abi);`

and gave me a Call Stack.(See image)

[image]

Also attaching the image of where it broke:
[image]

After I hit continue it gave me the same log as usual:

```
SensorVisualization.exe has triggered a breakpoint.

Exception thrown at 0x00007FF8E5F839EC (KernelBase.dll) in SensorVisualization.exe: WinRT originate error - 0x80070006 : 'The handle is invalid.'.
Exception thrown at 0x00007FF8E5F839EC in SensorVisualization.exe: Microsoft C++ exception: winrt::hresult_error at memory location 0x000000141DAFFA40.

onecoreuap\analog\input\holographicdriverclientlib\lib\spatialgraphdriverclient.cpp(2114)\Windows.Mirage.dll!00007FF8DC39BBF4: (caller: 00007FF8DC39BCA4) ReturnHr(1) tid(186c) 80070006 The handle is invalid.
    CallContext:[\ClientComponentManager_OnSensorInterestCallback\DestroySpatialGraphNodeReference] 
onecoreuap\analog\input\holographicdriverclientlib\lib\spatialgraphdriverclient.cpp(2141)\Windows.Mirage.dll!00007FF8DC39BCC4: (caller: 00007FF8DC39B2E0) ReturnHr(2) tid(186c) 80070006 The handle is invalid.
    CallContext:[\ClientComponentManager_OnSensorInterestCallback\DestroySpatialGraphNodeReference] 
SensorVisualization.exe has triggered a breakpoint.

Exception thrown at 0x00007FF8C105B5B8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.

SensorVisualization.exe has triggered a breakpoint.

Exception thrown at 0x00007FF8C105A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.

'SensorVisualization.exe' (Win32): Loaded 'B:\Windows\System32\DriverStore\FileRepository\qcdx_ffu850.inf_arm64_40919ccee4c23226\qcdx12arm64um850.dll'. 
Unhandled exception at 0x00007FF8C105A5D8 (ResearchModeAPI.dll) in SensorVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000010.

The program '[6312] SensorVisualization.exe' has exited with code 0 (0x0).

```

Hope this helps. Thanks.

aal-rvr, 2020-10-06:
Hi @dorinung , after the latest update(19041.1384) the crash is no longer taking place.
The AHAT frame not arriving still happens sometimes, always gets fixed after a reboot.
Thanks!

closed 2020-11-03 by kysucix