microsoft/HoloLens2ForCV #12: Strange depth value with AHAT

qian256, 2020-09-05:
Capturing AHAT depth frame using SensorRecording sample app -> converting into PLY -> visualize using Open3D, provides this results below (closer pixels is red). Notice the envelope-like area around the hand. It supposes to be a wall with some distance.
[image]
When visualized with the SensorVisualization sample app, it is also confirmed that the background wall in this region is detected to have similar depth as the hand. Is it possible to filter this area out?

qian256, 2020-09-05:
I found the description about "“aliased depth" in the arxiv paper, which basically offsets every depth pixel > 1m back into the 1m range. Any possibility to get back the actual depth value?

fbogo, 2020-09-09:
Correct, we tried to highlight the characteristics of depth in AHAT mode in both the API docs and the tech report to make it as clear as possible.
The mode uses single frequency, and it's not possible to run de-aliasing to compute depth for it. However, it should be possible to develop a method to identify "wrapped" depth pixels, at least invalidate them -- for example, leveraging DNNs and the active brightness images.
 
Some work going in this direction: http://www.audentia-gestion.fr/MICROSOFT/depth4free_SIGGRAPH.pdf

closed 2020-09-25 by kysucix