microsoft/HoloLens2ForCV #14: IMU sensors not supported?

max-krichenbauer, 2020-09-10:
Hello!

I'm trying to read the IMU (Acceleration, Gyro, Magnetometer) data on a Hololens two.
For this I extended the `StreamRecorder` sample to also handle the `IMU_ACCEL`, `IMU_GYRO`, and `IMU_MAG` sensor types.

Getting the sensor via `m_pSensorDevice->GetSensor(IMU_MAG, &m_pMAGSensor);` returns `S_OK` and a valid sensor reference;

But when `OpenStream()` is called on that sensor, then the returned result is `HRESULT_FROM_WIN32(ERROR_GEN_FAILURE) : A device attached to the system is not functioning.`.

Is this an hardware issue?
Is there some setting or initialization I'm missing?
I could not find anything in the `ResearchMode-ApiDoc.pdf` regarding differences in the initial set-up.

Thanks in advance!

By the way, in `ResearchMode-ApiDoc.pdf`, page 9 there seems to be a typo:

> Each sensor object defines sensor specific interfaces that can by QIed for.

fbogo, 2020-09-10:
I haven't much experience with IMU sensors, but have you checked section "Consent Prompt" from ResearchMode-ApiDoc.pdf? Using IMU requires an additional change to the app manifest.

dorinung, 2020-09-10:
To read IMUs the StreamRecorder needs to be modified to also request user consent for IMUs.  Please refer to code in SensorVisualizationScenario::IntializeSensors https://github.com/microsoft/HoloLens2ForCV/blob/main/Samples/SensorVisualization/SensorVisualization/SensorVisualizationScenario.cpp that uses RequestIMUAccessAsync. The same code needs to be added to SensorScenario::InitializeSensors in https://github.com/microsoft/HoloLens2ForCV/blob/main/Samples/StreamRecorder/StreamRecorderApp/SensorScenario.cpp

max-krichenbauer, 2020-09-11:
Thank you @fbogo and @dorinung !

Adding `<DeviceCapability Name="backgroundSpatialPerception"/>` to the manifest and `m_pSensorDeviceConsent->RequestIMUAccessAsync()` to the code solved the issue.

closed 2020-09-11 by max-krichenbauer

ozgunkaratas, 2020-12-09:
Hello, sorry to open up a closed topic but i wanted to ask you where you added the code to the manifest and to the SensorScenario.cpp because when i add _m_pSensorDeviceConsent->RequestIMUAccessAsync()_ it says "too few arguments in function call"

dakesse, 2020-12-11:
In the Package.appxmanifest you have to add it inside the 'Package'-tag and in the 'Capabilities'-block like this:
````
<?xml version="1.0" encoding="utf-8"?>
<Package ...  xmlns:rescap="http://schemas.microsoft.com/appx/manifest/foundation/windows10/restrictedcapabilities" IgnorableNamespaces="uap uap2 uap3 uap4 rescap mp mobile iot" ...>
...
  <Properties>
    ...
  </Properties>
  <Dependencies>
    ...
  </Dependencies>
  <Resources>
    ...
  </Resources>
  <Applications>
    ...
  </Applications>
  <Capabilities>   
    <rescap:Capability Name="perceptionSensorsExperimental" />
  </Capabilities>
</Package>
`````

ozgunkaratas, 2020-12-11:
Hey dakesse,
i was able to change the manifest file, but im still having trouble giving consent to the IMUs within the _InitalizeSensor_ block. i am looking at the SensorVisualizationApp to get some ideas but still no success.