microsoft/HoloLens2ForCV #20: In DEPTH_AHAT mode, the StreamRecorder program cannot work

jinpengxuan, 2020-09-16:
When setting the following configuration：
std::vector<ResearchModeSensorType> AppMain::kEnabledRMStreamTypes = { ResearchModeSensorType::DEPTH_AHAT };
The following logic does not work correctly：
       ```
     while (!pCameraReader->m_fExit && pCameraReader->m_pRMSensor)
        {
            HRESULT hr = S_OK;
            IResearchModeSensorFrame* pSensorFrame = nullptr;

            hr = pCameraReader->m_pRMSensor->GetNextBuffer(&pSensorFrame);

            if (SUCCEEDED(hr))
            {
                std::lock_guard<std::mutex> guard(pCameraReader->m_sensorFrameMutex);
                if (pCameraReader->m_pSensorFrame)
                {
                    pCameraReader->m_pSensorFrame->Release();
                }
                pCameraReader->m_pSensorFrame = pSensorFrame;
            }
        }
```

 The GetNextBuffer function crashed, resulting in the inability to perform the next step

fbogo, 2020-09-16:
Have you observed this behavior since the first time you deployed the app?
There is a known issue with AHAT depth that needs further investigating and is in our radar.
Could you try to reboot the HoloLens and see if the problem remains?

naleksiev, 2020-09-18:
I had the same issue. This is the only project that has ARM (32bit) target platform, which happens to be the default one. For fix check PR #23

closed 2020-09-21 by kysucix

chiarapalu, 2023-02-09:
I'm having the same issue while running the app in AHAT mode. I Installed it again after a few months (I was using it without any problem a few months ago).
Any help? Thanks!