microsoft/HoloLens2ForCV #22: ARM Support

vos90, 2020-09-17:
Hello,

Is there any way to compile samples in ARM configuration (not ARM64)?

kysucix, 2020-09-17:
hi vos90,

no that's not possible because hololens 2 includes a[ Snapdragon 850](https://www.qualcomm.com/products/snapdragon-850-mobile-compute-platform) which has a 64-bit architecture.

Silvano

dorinung, 2020-09-17:
We are working on adding OS support for running RM apps compiled for ARM on the OS.

dorinung, 2020-09-17:
The samples can be compiled for ARM. However they will not run correctly at the moment. This was also reported here: https://github.com/microsoft/HoloLens2ForCV/issues/11

eshel, 2020-10-06:
@kysucix / @dorinung - could you clarify why ARM32 cannot be supported properly?

In our project, we rely on another library for Hololens 2 - [MixedReality-WebRTC](https://github.com/microsoft/MixedReality-WebRTC), which specifically cannot compile on ARM64 yet (adding support there is a massive undertaking, as described in 
[mrwebrtc#414](https://github.com/microsoft/MixedReality-WebRTC/issues/414) due to build complexities in the underlying webrtc library).

This leaves us in the awkward spot of having to choose - either target ARM32 and have WebRTC, or target ARM64 and have Research Mode, but not both simultaneously.

Is it somehow possible to mix & match ARM32 and ARM64 DLLs in the same runtime (in our case it happens to be a unity app)? Alternatively, what does supporting research mode in ARM(32) involve? 

Any help will be very appreciated!
Thanks in advance

dorinung, 2020-10-06:
We are working on providing arm32 support. arm32 and arm64 can not be used in the same executable. Will have an update in the next few weeks.

whirwind, 2020-11-17:
Hello, same problem here.

I want to get RGB video stream using [MixedReality-WebRTC](https://github.com/microsoft/MixedReality-WebRTC/tree/master), then get IMU and Depth data using RM api windows runtime component plugin. All project is done in Unity.

One of them can only work in ARM and another one can only work in ARM64, this is really ……