microsoft/HoloLens2ForCV #24: Occur Crush in InitializeSensors()

forarete, 2020-09-21:
Hello,

I'm newbee on hololense developer, and thanks your contributions.

I have troubled to run sample projects.

The sample 'CalibrationVisualization' is OK. But the others is not worked.

When I tried debugging, The error is occur on follows line.

 winrt::check_hresult(m_pSensorDevice->QueryInterface(IID_PPV_ARGS(&m_pSensorDeviceConsent)));

[image]

I already allowed access to camera stream in [hololense2 setting/privacy] and [device portal/research mode].

dorinung, 2020-09-21:
What version of the OS do you have on Hololens? Thanks.

forarete, 2020-09-22:
It's

Windows version	19041.1117.arm64fre.vb_release_svc_sydney_rel_prod.200903-1632

Thanks.

dorinung, 2020-09-22:
The earliest build that fully supports research mode is 19041.1364. Please join the Windows Insider Program to get preview builds. After that, in the device portal, enable research mode, different than recording mode. See https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ECCV2020-Tutorial/ECCV2020-ResearchMode-Api.pdf (slides 6, 7 and 8) or Setup section in https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ResearchMode-ApiDoc.pdf.

forarete, 2020-09-23:
It works after updating preview build!

I thought my OS was up to date because I updated it in settings. But it was not preview build version you mentioned.

Sorry for confusing you.

Thanks!

closed 2020-09-23 by fbogo