microsoft/HoloLens2ForCV #27: SensorVisualization  Project operation problem

RenSiyao, 2020-09-23:
When I run the program, it will only show the application window. What is the situation? I hope someone can answer for me.
I checked the logs in the background and found no.
  
[image]

rooi, 2020-09-23:
Yesterday I ran into the same issue. In my case research mode was disabled in the device's portal. Enabling the checkbox and restarting the device solved it in my case

RenSiyao, 2020-09-23:
> 昨天我遇到了同样的问题。在我的情况下，设备门户中的研究模式已禁用。启用复选框并重新启动设备就我而言解决了它
My device is in researcher mode, and I wonder if it's because my release goal is 10.0.18362.0.
What is your release version?

dorinung, 2020-09-23:
Please check the following:
	- OS on Hololens is 19041.1364 or later. Please join the Windows Insider Program to get preview builds,
	- in the device portal, enable research mode, different than recording mode. 

For setup details See https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ECCV2020-Tutorial/ECCV2020-ResearchMode-Api.pdf (slides 6, 7 and 8) or Setup section in https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ResearchMode-ApiDoc.pdf.

RenSiyao, 2020-09-24:
> 请检查以下各项：
> -Hololens上的操作系统为19041.1364或更高版本。请加入Windows Insider计划以获取预览版本，
> -在设备门户中，启用不同于录制模式的研究模式。
> 
> 有关设置的详细信息，请参见  [https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ECCV2020-Tutorial/ECCV2020-ResearchMode-Api.pdf（幻灯片](https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ECCV2020-Tutorial/ECCV2020-ResearchMode-Api.pdf%C2%A0(slides) 6、7和8）或[https：// github中的``](https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ResearchMode-ApiDoc.pdf)设置''部分  [.com / microsoft / HoloLens2ForCV / blob / main / Docs / ResearchMode-ApiDoc.pdf](https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ResearchMode-ApiDoc.pdf)。

Here are my screenshots of Hololens2 preview Settings, Hololens2 system version, and Researcher mode screenshots.
[image]
 
[image]

[image]

Even if my project is still not successful, what else should I pay attention to?I hope you can help me.
[image]

Lancename, 2020-09-24:
You'll need to participate in the Microsoft experience program, then update the system to a private beta version (both on your Hololens2 device) and then open researcher mode in your computer's browser
你需要加入微软的体验计划， 然后更新系统，更新到内测版本（都是在Hololens2设备里面)最后才能在电脑的浏览器里面打开研究者模式
[HoloLens2初入——打开研究者模式（Research mode）](https://blog.csdn.net/scy261983626/article/details/108754517)

dorinung, 2020-09-24:
Your screen shot above shows recording mode. Please find and enable research mode.

RenSiyao, 2020-09-25:
> 您需要参与Microsoft体验计划，然后将系统更新到私有Beta版本（在Hololens2设备上），然后在计算机的浏览器中打开研究员模式。您
> 需要加入微软的体验计划，然后更新系统，更新到内测版本（都是在Hololens2设备里面）最后才能在电脑的浏览器里面打开研究者模式
> [HoloLens2初入-打开研究者模式（研究模式）](https://blog.csdn.net/scy261983626/article/details/108754517)

Thanks for your help, my project is ready to run.

RenSiyao, 2020-09-25:
> 上面的屏幕快照显示了录制模式。请找到并启用研究模式。

Thanks for your help, my project is ready to run.

closed 2020-09-25 by dorinung