microsoft/HoloLens2ForCV #28: IMU Framerate

dakesse, 2020-09-24:
In which frame rate can the IMU sensor data be retrieved?
As i understand, via polling (with _GetNextBuffer()_) you can get the first sensor value of an IMU frame (_GetCalibratedAccelaration()_) or all of them batched (_GetCalibratedAccelarationSamples()_). however, the IMU frames themselve are limited to a certain rate, which makes this access method not particularly suitable for realtime data processing.  I figured out 12fps for the accel and around 24fps for the gyro.
Is there also an event based approach? So that you get values ​​as soon as new ones are available from the sensor.

dorinung, 2020-09-24:
Your fps estimates are correct. Sensor reading can not be retrieved one by one, On Hololens2 IMU sensors deliver readings in batches. As far as event use you could have dedicated worker thread for each sensor that sits in loop reading batches. On batch arrival the batch could be placed in a work queue that has an associated work event that is signaled. The queue could be read by a different processing thread.

Hope this helps.

dakesse, 2020-09-28:
Thank you, that helps a bit. But to be sure:
It is not possible to get sensor data simultaneously with a respective render frame?

dorinung, 2020-10-16:
Due to hardware architecture each sensor generates frames at its own cadence and they can not be synchronized with an external timeline. They can not be synchronized with other events/threads.

dakesse, 2020-10-16:
Thank you for the further explanation.

closed 2020-10-16 by dakesse