microsoft/HoloLens2ForCV #30: Interpreting magnetometer data

max-krichenbauer, 2020-09-25:
Hello!

I'm reading the magnetometer values from the Hololens 2, but I can't figure out what they are supposed to mean or if my readings are erroneous.
The documentation doesn't state any unit of measurement or meaning for these values.

When I tested it, the values where [464, -198, 483] and after turning  180 degrees they were [456, -223, 463].
What do these values mean an how are they to be processed.

Thank you!

Best regards,
Max

ozgunkaratas, 2020-12-11:
Hey Max,
have you had any update regarding data extraction from the IMU sensor?

max-krichenbauer, 2020-12-15:
Hi @ozgunkaratas 
Through other channels, I received the following explanation:

```
They *think* the units for the returned values are micro-teslas.
[...]
I'm guessing that the HL2 only uses the magnetometer to sense relative heading. So it's probably not calibrated to do more than that. For example, the research API does not give any extrinsic calibration data on how the magnetometer is precisely oriented to the coordinate frames of the other sensors or cameras. This would also explain why they aren't 100% sure of the units – because they might only be looking at how the direction of the magnetic field changes with HL2 orientation (and not using the absolute field strength).
```

ozgunkaratas, 2020-12-29:
Hi @max-krichenbauer 
thank you for your update, i was able to compare HL magnetometer data with my external IMU sensor which outputs field strength information in 3 axes in units of Tesla and the HL and the IMU do not match with each other whatsoever. I think you are right in the sense that this is used for sensing relative heading.

made3, 2021-04-20:
Hey @max-krichenbauer 
Did you find a way to get a degree value or something like that from the magnetometer outputs? It is hard to find anything on the internet about using the magnetometer values of the HoloLens...

ilyassebel, 2021-09-27:
Hi 

> Hello!
>
> I'm reading the magnetometer values from the Hololens 2, but I can't figure out what they are supposed to mean or if my readings are erroneous.
> The documentation doesn't state any unit of [...]

Hi Max, 

Please how did you get the magnetometer values ? 

Best regards