microsoft/HoloLens2ForCV #34: how to get 4 tracking cameras  intrinsics and distort imgs buffer

buqing2009, 2020-09-29:
when i use **StreamRecorder** to record sensors data, i found that the saved imgs of 4 tracking cameras are undistorted. 
How to get raw data of 4 tracking cameras and these intrinsics?

dorinung, 2020-09-29:
The images are raw camera images. My captures show perspective distortion. The intrinsic are not directly exposed by the API. They are factored into MapImagePointToCameraUnitPlane methods.

[image]

buqing2009, 2020-10-09:
here is the chessboard image from two front tracking cameras,
[image]
[image]

dorinung, 2020-10-16:
The camera intrinsics are not exposed by the API. Instead a MapImagePointToCameraUnitPlane  method for each camera is provided. The camera intriniscs model is used to implement this method. This can used to project image points to camera unit vectors. A data set of vectors, image points pairs can then be used to fit a camera model.