microsoft/HoloLens2ForCV #37: Access Camera Intrinsics

bakader, 2020-10-02:
For many OpenCV functions I need to input the Intrinsics of the cameras. However they are not directly accessible as written in the HoloLens2 API paper. Could you please provide some Information on how to access the camera Intrinsics of the HoloLens2 cameras?

kysucix, 2020-10-02:
hi,

intrinsic for depth and gray scale camera are not provided directly. What is provided is a function (or a lookup table if you're using the recorder app) from pixel to unit ray in the camera space.

kysucix, 2020-10-02:
The right function on the device is MapImagePointToCameraUnitPlane()

bakader, 2020-10-02:
Hi,
Thank you for the fast response. I am aware of the mentioned function however i do need the intrinsic parameters for some OpenCV functions. Is there a way to determine these intrinsic parameters by the MapImagePointToCameraUnitPlane() function? Especially since the mentioned function only maps 2D-->2D (i.e. xy to uv and vice versa). I am kind of missing a 3D->2D mapping in order to determine the intrinsic parameters.

dorinung, 2020-10-16:
The values returned MapImagePointToCameraUnitPlane returns the xy coordinates of a unit plane vector. z is 1. Please see usage of MapImagePointToCameraUnitPlane in https://github.com/microsoft/HoloLens2ForCV/blob/main/Samples/CalibrationVisualization/CalibrationVisualization/CalibrationVisualizationScenario.cpp: 

        uv[0] = 640.0f;
        uv[1] = 0.0f;

        pCameraSensor->MapImagePointToCameraUnitPlane(uv, xy);

        vectorOriginRenderer = std::make_shared<VectorModel>(m_deviceResources, 0.1f, 0.0005f, DirectX::XMFLOAT3(xy[0], xy[1], **1.0f**));

kysucix, 2020-11-03:
@bakader you would need to get the unit camera ray and project them to a virtual pinhole camera with the intrinsic parameters that you're choosing.

closed 2020-11-03 by kysucix

Bornblack, 2021-09-17:
hi!  I@ kysucix 
Are the values in lookup table of the depth camera the same at different times? or rather the instrinsic of the camera is constant? In the StreamRecorder example, the lookup table is saved at the end of the recording once. If the instrinsic is not constant, will it affect the accuracy of the point cloud reconstructed from the depth image ? 

Any reply would be appreciated.

evia100, 2022-05-16:
> hi! I@ kysucix Are the values in lookup table of the depth camera the same at different times? or rather the instrinsic of the camera is constant? In the StreamRecorder example, the lookup table is [...]

Hi , did you managed to understand if the LUT  is constant over time?

chiarapalu, 2023-02-23:
Did someone understand how to get the intrinsic parameters for the depth camera? 
Thanks a lot!

siatheindochinese, 2023-04-30:
I managed to approximate the intrinsic parameters through projective geometry by using all the pixel-ray correspondences available in the lookup table provided. You can check out my results here: [LUT_to_K](https://github.com/siatheindochinese/LUT_to_K).

The depth camera is a bit tricky, visually plotting the rays show that the focal lengths are really, really small.