microsoft/HoloLens2ForCV #38: DEPTH_AHAT Frames arriving at a max of 10 fps

aal-rvr, 2020-10-07:
Hi,

The API doc states that the DEPTH_AHAT sensor runs at 45 fps. We are trying to figure out the actual frame rate that we receive.
We found that the time difference between each frame is 0.2 to 0.1 seconds.
I just took the difference between the SensorTicks and divided them by SensorTicksPerSocond to get that.
The difference between System Time for the frames is consistent with the SensorTicks.
I tested this in CameraUpdateThread in the SensorVisualization Sample.
Did I miss something or is that the best framerate for DEPTH_AHAT?

Thanks.

fbogo, 2020-10-08:
In general, the framerate might depend on how many streams you open, if you're in Debug or Release etc..

If you try the StreamRecorder app, when you extract the recorded data via `recorder_console`, the code will call the `check_framerates` function from `utils.py` and you should see the framerate printed out.
Perhaps you could just check which value you get there.

aal-rvr, 2020-10-09:
Hi @fbogo ,

Thanks for the suggestion. I ran the `StreamRecorder` app and `check_framerates` prints out:
`Average Depth AHaT delta: 460.476ms, fps: 2.172`
I have been running the `SensorVisualiztion` sample program in Release configuration, with only AHaT stream open and all  frame processing disabled.
After running the `StreamRecorder` I modified the `SensorVisualiztion` app to use the the same time conversion function from `StreamRecorder` app - `RelativeTicksToAbsoluteTicks()` in the `TimeConverter` class and got the deltas between the frames from 130 to 200 milliseconds and is consistent with system time called when the frame arrives.
I also checked that `DisableEyeSelection()` does get called in the `SensorVisualization` app, so the framerate must be the highest possible.
I'm trying to synchronize the depth frames with other devices, which is why I need the timestamps during runtime. Is there something I'm missing?
Why is it that I'm only getting a max of 5-7 fps, with no frame processing performed?

Thanks.

fbogo, 2020-10-12:
Hmm looks like I cannot repro (I'm getting almost 45fps).
Just to check, what's the framerate you get from `recorder_console` for hand/head tracking? (should get printed together with AHAT framerate).

When you run `SensorVisualization`, is the visualization of AHAT clearly at such a low framerate?

aal-rvr, 2020-10-13:
I ran the StreamRecorder app again just to be sure. I don't know if there was some issue with the build the last week (I am pretty sure I was running in Release mode), but I couldn't get more than 2fps.
Today I got values from 25fps to 29fps for the AHAT stream and hand/head tracking was consistently ~59 fps
This is the best one so far:
```
Average Depth AHaT delta: 34.231ms, fps: 29.213
Average hand/head delta: 16.752ms, fps: 59.693
```

In Release mode, printing out Debug logs causes the framerate to slow down to 1 to 2 fps.
In the StreamRecorder App, in Release - I stored the deltas between frames and checked it after 100 frames and there it seems to be getting close to 45 fps every time.

The AHAT visualization is sometimes visibly low (especially in Debug config).
So. right now it looks like I am getting a high enough framerate that we can use.

Thank you so much for your help! :)

closed 2020-10-13 by fbogo