microsoft/HoloLens2ForCV #40: Package the project into DLL libraries

RenSiyao, 2020-10-14:
How to put the project into Unity after packaging the DLL library。
Who can tell me the complete process of packaging a project into a DLL library and how to call this DLL inside Unity?
I used the call method shown in the figure below but got an error in Unity.Who can help me?
[image]
[image]

dakesse, 2020-10-14:
Have you tried using `DUMPBIN /exports my.dll` to get the correct entry points?
This gives you something like this:
```
[DllImport("my.dll", EntryPoint = "?InitSensors@DllExports@@SAXXZ")]
public static extern void initSensors();
```

RenSiyao, 2020-10-19:
> 您是否尝试过使用`DUMPBIN /exports my.dll`以获取正确的入口点？
> 这给你这样的东西：
> 
> ```
> [DllImport("my.dll", EntryPoint = "?InitSensors@DllExports@@SAXXZ")]
> public static extern void initSensors();
> ```
First of all, thank you for your reply. I still cannot detect the method in DLL library according to the method you provided.Do you also try to use sensor data from Hololens2ForCV in Unity for your own projects?Can you tell me the complete process?From wordpress posts to Holoelns and how to build DLL libraries.

dakesse, 2020-10-21:
> > 您是否尝试过使用`DUMPBIN /exports my.dll`以获取正确的入口点？
> > 这给你这样的东西：
> > ```
> > [DllImport("my.dll", EntryPoint = "?InitSensors@DllExports@@SAXXZ")]
> > public static extern void initSensors();
> > ```
> 
> First of all, thank you for your reply. I still cannot detect the method in DLL library according to the method you provided.Do you also try to use sensor data from Hololens2ForCV in Unity for your own projects?Can you tell me the complete process?From publishing projects to Hololens and how to build DLL library files.

Hi @RenSiyao ,
yes I use research mode sensor data in Unity, strictly speaking accelerometer and gyro values.
If you haven't already, take a look into #31.

The complete process would be a bit to much but here are the general steps:

- For the DLL I used a VS 2019 C++ DLL project in which I added the research mode api header. then I implemented functions to init the sensors and get their data analogue to the [SensorVisualization](https://github.com/microsoft/HoloLens2ForCV/tree/main/Samples/SensorVisualization) example. These are my functions declared in the DLL's header:

```
class DllExports {
public:
  DllExport static void InitSensors();
  DllExport static void OpenAccSensorStream();
  DllExport static void OpenGyroSensorStream();
  DllExport static void OpenMagnetSensorStream();

  DllExport static void GetAccDataStamped(float& accX, float& accY, float& accZ, UINT64& timestamp);
  DllExport static void GetGyroDataStamped(float& gyroX, float& gyroY, float& gyroZ, UINT64& timestamp);
  DllExport static void GetMagnetDataStamped(float& magnetX, float& magnetY, float& magnetZ, UINT64& timestamp);
  DllExport static SensorDataStruct* GetAccDataBatched();
  DllExport static SensorDataStruct* GetGyroDataBatched();
  DllExport static SensorDataStruct* GetMagnetDataBatched();

  DllExport static void CloseAccSensorStream();
  DllExport static void CloseGyroSensorStream();
  DllExport static void CloseMagnetSensorStream();
  DllExport static void ReleaseSensors();
};
```
Target platform has to be ARM64.

- I tested my DLL with a C++ UWP app before including it to Unity.
- For Unity I needed the correct entry points of my exported functions. Via the developer command prompt for VS and the command `dumpbin /exports my.dll` you can get them.
- Then copy the DLL into your Unity project and import the functions that you want to use in a script:

```
[DllImport("my.dll", EntryPoint = "?InitSensors@DllExports@@SAXXZ")]
public static extern void initSensors();
...
```
Here you have to be careful how you pass data from C++ to C#. I solved it with an 'unsafe' block but marshalling can be a way too.
- After the first build you have to modify the Package.appxmanifest as discribet in the [ResearchMode API Doc](https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ResearchMode-ApiDoc.pdf) and add `<rescap:Capability Name="perceptionSensorsExperimental" />`.
- Research mode must be activated in the Device Portal as described [here](https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ECCV2020-Tutorial/ECCV2020-ResearchMode-Api.pdf).
- Finally you can deploy to your HoloLens 2.

I hope that helps a little. Otherwise just feel free to ask.

agroenenberg, 2020-10-26:
@RenSiyao I'm trying to get the StartRecording() and StopRecording() functions incorporated as well but can't get the winrt::Windows library in my DLL. How did you manage to do that? Did you use a Windows Runtime Component format? Thanks, it would help me a lot

RenSiyao, 2020-10-26:
> @RenSiyao我正在尝试同时合并StartRecording（）和StopRecording（）函数，但是无法在我的DLL中获得winrt :: Windows库。您是如何做到的？您是否使用Windows运行时组件格式？谢谢，对我有很大帮助
It's a pity that I can't help you, because my project hasn't been built successfully yet.

agroenenberg, 2020-10-26:
Your DLL built succesfully right? Did you look at #31 ? I had the same problem and solved it. Did you put your DLL in your Assets/Plugins folder?

RenSiyao, 2020-10-26:
> 您的DLL成功建立了，对吗？你看到[＃31了](https://github.com/microsoft/HoloLens2ForCV/issues/31)吗？我有同样的问题并解决了。您是否将DLL放在了Assets / Plugins文件夹中？

The DLL library I successfully built is a case study in the depth camera project, and my DLL library cannot be used on the journey.
I used the DLL library built into this project.Hope I can help you.[https://github.com/petergu684/HoloLens2-ResearchMode-Unity](url)

agroenenberg, 2020-10-26:
I'll look into it and let you know if I can get it to work!

RenSiyao, 2020-10-26:
You can only get one depth sensor data for the DLL library I provided to you. I cannot solve your function problem of StartRecording () and StopRecording () for you.

JiangTao100, 2021-02-10:
@RenSiyao 
Your page: https://github.com/petergu684/HoloLens2-ResearchMode-Unity has disappeard.

JiangTao100, 2021-02-10:
@dakesse  Today I build my own dll and use the plugin in Unity ,when I wanna modify Package.appxmanifest I find  I cannot  add<rescap:Capability Name="perceptionSensorsExperimental" />,the error showed that "undefined namespace prefix ".Have you faced this problem?

JiangTao100, 2021-02-10:
I find rescap here :https://docs.microsoft.com/en-us/windows/uwp/packaging/app-capability-declarations