microsoft/HoloLens2ForCV #41: Problem building and deploying the different solution files

mastericoon, 2020-10-14:
Hi,
I have some errors building and deploying the different solution files. One error is instructing to reinstall Windows, but I first want to verify whether you have other suggestions? Thank you in advance for your advice.
Best regards,
kv

**CameraWithCVAndCalibration.sln**
	
Following error occurred during building:

[image]

This error refers to the following error code: LNK1107
https://docs.microsoft.com/en-us/cpp/error-messages/tool-errors/linker-tools-error-lnk1107?f1url=%3FappId%3DDev16IDEF1%26l%3DEN-US%26k%3Dk(LNK1107)%26rd%3Dtrue&view=vs-2019
I thought it might be related to the use of OpenCV.

**SensorVisualization.sln**
Building succeeded

During Deploy the following error message occurred:

‘Unable to activate Windows Store app ‘…SensorVisualization’.
A Microsoft Windows component is not correctly registered. If the problem persists, try repairing your Windows installation, or reinstalling Windows.
[image]

**StreamRecorder.sln**
Building succeeded
During Deploy the same error as for SensorVisualization.sln occurred.

[image]

mastericoon, 2020-10-16:
Yes, I thought that too, but unfortunately that didn't solve the problem. Developer mode is enabled.

maguileracanon, 2020-10-16:
I am having the same issues with SensorVisualization and StreamRecorder. I repaired VS just to be sure, I know I have developer mode / recording mode enabled and already installed the latest OS build from the insiders program.

dorinung, 2020-10-16:
What version of the OS is installed on your Hololens?

dorinung, 2020-10-16:
To make sure VS is properly configured could you please try building and running the DirectX 11 holographic UWP app template. Instructions are at: https://docs.microsoft.com/en-us/windows/mixed-reality/develop/native/creating-a-holographic-directx-project .

mastericoon, 2020-10-16:
OS build is 10.0.1941.1117, is that ok?
I also have the following errors, when building the DirectX 11 holographic UWP app:
[image]

Should I configure VS differently now?

maguileracanon, 2020-10-19:
[image]
Yes, same with me, and I am using 19041.1388

maguileracanon, 2020-10-19:
@mastericoon I think I managed to solve this. I hope my solution works for you. First I swiched to the Beta Channel of the developers insider program. Then I could see the "Research Mode" in the developers portal. I thought that *Reording Mode* was the same as *Research Mode* in Hololens2. (They are easy to confuse given the pictures available in the tutorial, also I read they were the same in the HoloDevelopers Slack group).
After finally seeing an option to enable Research Mode I was able to run the SensorVisualization and StreamRecorder apps. I hope this helps! :D

mastericoon, 2020-10-19:
Ok, thanks. Indeed, confusing that 'Recording Mode' and 'Research Mode' look so similar. I also was confused by this.
I also just saw that I need to use at least build 1356 as explained in the tutorial:
'Research Mode for HoloLens 2 is available beginning with build 19041.1356. If you need access in an earlier build, sign up for our Insider Preview program.'

closed 2020-10-21 by mastericoon