microsoft/HoloLens2ForCV #45: Question about IMU Timestamp

mianbao96, 2020-10-29:
Refer to the acquisition method given in  the API document, I have successfully accessed the IMU data，but I found a question about timestamp. For IMU sensor，there are two kind of timestamps, which are in HostTicks (hundred of nanoseconds) and SensorTicks (nanaseconds). According to the collected data, the SensorTicks in nanoseconds grows at equal intervals, even between different sensor frames, but HostTicks in hundreds of nanaseconds of all samples in each sensor frame are the same, I want to ask if this is normal? Why it came out like this? And how can I get the accurate timestamp in HostTicks of each sample？

dorinung, 2020-11-04:
There is a bug where all socticks for a batch are the same as the soc tick for the first sample in the batch.

To get SocTicks for all samples please use the VinylHupTicks fields of the imu samples to calculate deltas from the first sample for each sample in the batch.

mianbao96, 2020-11-06:
@dorinung Thank for your reply. Actually when I discovered this problem, I did the same. For each IMU frame, I take the common SocTicks obtained in all samples as the first sample's SocTicks, and them use the VinyHupTick to calculate the exact deltas between the continuous samples, thus I can get the SocTicks for all samples in the frame. But I found a new problem. If you follow this method, sometimes the calculated last few samples' SocTicks of the previous IMU frame is larger than the first sample's SocTicks of the next IMU frame. Obviously this is wrong, It means that the common SocTicks of all samples within an IMU frame is not the exact SocTicks of the first sample. I'm looking forward to your answer, thank you.

mianbao96, 2020-11-25:
@dorinung  I'm really sorry to bother you again. I would like to ask if the bug about the SocTicks will be resolved? Or will you give a better method to calculate it? Please reply to my questions when you are free. Thank you very much.

dorinung, 2020-11-25:
Sorry for the delayed response. I am in the process of debugging the issue and maybe providing a workaround.

JiangTao100, 2021-01-30:
Hi,mianbao96. I'm kind of new to the IMU API.After I  read the API document, I still cannot access  the IMU data. I will appreciate much if you give  some codes  for newers.Thanks a lot!

dakesse, 2021-02-01:
@JiangTao100 Have a look at [Samples/SensorVisualization/](https://github.com/microsoft/HoloLens2ForCV/tree/main/Samples/SensorVisualization).

wystephen, 2021-03-02:
> Sorry for the delayed response. I am in the process of debugging the issue and maybe providing a workaround.

Hi, is there any solution to calculate the right timestamp for each IMU measurement?
Furthermore, how about the accuracy of the synchronization of the camera and IMU?

zijianh4, 2024-06-29:
Hi,
Has anyone solve this issue? It seems hard to interpret the soctick of IMU data or do some post processing to get approximately correct ones.