microsoft/HoloLens2ForCV #46: Start/stop button doesn't function

ishanic, 2020-10-30:
Hello, I am unable to interact with the start/stop button. I can see the buttons when I turn on the app, but can't "press" it. Could you help? Thanks.

weders, 2020-10-30:
Initially, I also had the feeling that it does not work. But when you press the 'START' button, you should notice a slight change in color of both buttons. Then, the recording started. When you are finished with recording, you can press the 'STOP' button and you should see the files in your file explorer.

fbogo, 2020-11-02:
@ishanic , could you please let us know if the problem is solved?

ishanic, 2020-11-03:
@fbogo the problem has not been resolved yet. I understand what @weders meant, but I see no color change or haptic/audio feedback at all, so it appears to be totally unresponsive. Is that something that can be debugged or is there a workaround?

kysucix, 2020-11-03:
@ishanic there was a workaround for a bug that appeared on Visual Studio which was addressed on a later release. I reverted the workaround recently but probably it's not safe to assume that everyone is running latest Visual Studio version so I reverted it in #48 .

closed 2020-11-03 by kysucix

ishanic, 2020-11-03:
Yes, the buttons are functioning as expected after this fix. Thank you!