microsoft/HoloLens2ForCV #49: Is there any plan to support machine learning on holoLens2

IkbeomJeon, 2020-11-03:
Hello, 

Do you have any existing tries or plan to support machine learning library on Its GPU in HoloLens2 ?

Recent days, I have tried to run onnx model in HoloLens2. It was successful to build and run onnxruntime(or WinML) library using Its CPU only (ARM architecture). But to use its GPU, these library require GPU providers (for exmaple, CUDA, DirectML, OpenCL) for data binding.

The GPU of Hololens2 is 'Qualcomm Adreano 630'. And according to follow documentation, it support OpenCL 2.0, and DX12 and so on. (see : https://www.qualcomm.com/products/snapdragon-850-mobile-compute-platform)

So I have been looking for a way to use these API in ML libraries. In onnxruntime and WinML , It is possible to use 'DirectML', which use dx12. But unfortunately, it was not support ARM but only x64, x86. (So it seems possible to run only hololens 1)

So I want to know Is there any idea or plan to run any ML library on HoloLen2 using Its GPU.

Thanks.

The spec. of holoLens2 is shown in this link.
https://build5nines.com/microsoft-hololens-2-hardware-specs-vs-hololens-1/

I always thanks to contributions.

kysucix, 2020-11-04:
hi,

did you try to use use official qualcomm library that support onnx inference?

IkbeomJeon, 2020-11-05:
No. I tried just to build 'DirectML Execution Provider' in onnxruntime.(https://github.com/microsoft/onnxruntime/blob/master/docs/execution_providers/DirectML-ExecutionProvider.md)
I thought that it will be worked if it can be build on ARM architecture. But the build configuration(onnxruntime  + dml provider) can be supported only in x86 or x64 now.

zc-alexfan, 2021-04-21:
I have a similar question. For example, I have scene understanding models written in PyTorch, but I am not sure how hard it is to apply that model on HoloLens for realtime inference.