microsoft/HoloLens2ForCV #51: Possibility to reduce framerate?

fehmerli, 2020-11-17:
Hello,

I investigate on the pre-heating-time of the HoloLens2, so I need to run the StreamRecorderApp for about 90 minutes. Unfortunately I noticed that the App runs very instable when using it for 10 or 15 minutes. After hitting the "Stop"-Button, the App takes several minutes to reload and it does not always give reliable results. Sometimes the app just crashes down without saving the recordings.

I suspect that this happens because of the high framerate (5fps in release-mode), so the app collects a big amount of data in that period of time.. Is there any possibility to reduce the framerate to make longer investigations possible (eg. 1 fps or still fewer)?

dorinung, 2020-11-23:
The API does not allow for controlling the sensor frame rates. You can try to modify the app recording loops to drop frames or use sleeps to slow down the capture rate.

fbogo, 2020-11-25:
Yes, one way to reduce the framerate inside the app could be to modify the `IsNewTimestamp` function in 
[RMCameraReader](https://github.com/microsoft/HoloLens2ForCV/blob/main/Samples/StreamRecorder/StreamRecorderApp/RMCameraReader.cpp) into a function that checks if the difference in timestamps is greater than a given threshold.

In general, the StreamRecorder app offers a demo about capture capabilities, but for specific used it should be customized and optimized. For instances, hand tracking data is saved in main memory and just dumped to disk only at the end of each capture sequence. This might be problematic for very long captures.

cgsaxner, 2020-12-19:
Hi @fehmerli,
in case you are still interested, i modified the `IsNewTimestamp` function to enforce a minimal difference of `minDelta` (in Hundreds of Nanoseconds) between frames. Maybe it is useful to you:
https://github.com/cgsaxner/HoloLens2-Unity-ResearchModeStreamer/blob/6e3d2ef3c6eec6fa875df70cc37d1ecce8a1f3fb/HL2RmStreamUnityPlugin/ResearchModeFrameProcessor.cpp#L175