microsoft/HoloLens2ForCV #53: Compiling error

harukary, 2020-11-23:
Hello, I would like to know how to run the demos.
I just downloaded this repo and opened the .sln file. Then, compiled with Debug mode for ARM64 over USB.
I got these errors.

identifier "await" is undefined	SensorVisualization	C:\HoloLens\HoloLens2ForCV-main\Samples\SensorVisualization\SensorVisualization\Content\ModelRenderer.cpp	271

'GyroUpdateThread': illegal qualified name in member declaration (compiling source file Content\GyroRenderer.cpp)	SensorVisualization	C:\HoloLens\HoloLens2ForCV-main\Samples\SensorVisualization\SensorVisualization\Content\GyroRenderer.h	55

'MagUpdateThread': illegal qualified name in member declaration (compiling source file Content\MagRenderer.cpp)	SensorVisualization	C:\HoloLens\HoloLens2ForCV-main\Samples\SensorVisualization\SensorVisualization\Content\MagRenderer.h	55

'AccelUpdateThread': illegal qualified name in member declaration (compiling source file Content\AccelRenderer.cpp)	SensorVisualization	C:\HoloLens\HoloLens2ForCV-main\Samples\SensorVisualization\SensorVisualization\Content\AccelRenderer.h	55

'AccelUpdateThread': illegal qualified name in member declaration (compiling source file AppView.cpp)	SensorVisualization	C:\HoloLens\HoloLens2ForCV-main\Samples\SensorVisualization\SensorVisualization\Content\AccelRenderer.h	55

I googled 'await', but nothing found. The other errors are removed by removing '&' from 'SetColor(DirectX::XMFLOAT3 &color)'
Would you tell me how to solve this problem?

dorinung, 2020-11-23:
Please try replacing await with co_await.

dorinung, 2020-12-11:
I have updated the CalibrationVisualization and SensorVisualization for compile errors that came with the latest version of Visual Studio 16.8. Can you please try the latest changes?

Thanks.