microsoft/HoloLens2ForCV #54: Exception "hresult_error" when debugging

ranchenhao, 2020-11-27:
I have successfully built the sample solution. But got "hresult_error" when debugging it on Hololens. 

Exception thrown at 0x00007FFDB5CC39EC (KernelBase.dll) in CalibrationVisualization.exe: WinRT originate error - 0x8000FFFF : 'Catastrophic failure'.
Exception thrown at 0x00007FF6833EF284 in CalibrationVisualization.exe: 0xC0000005: Access violation reading location 0x0000000000000000.

The first exception occurred at "winrt\base.h"
`throw hresult_error(result, take_ownership_from_abi);`

The second exception occurred at "Windows.UI.Core.h"
`template <typename D> WINRT_IMPL_AUTO(void) consume_Windows_UI_Core_ICoreWindow<D>::KeyDown(winrt::event_token const& cookie) const noexcept
    {
        WINRT_VERIFY_(0, WINRT_IMPL_SHIM(Windows::UI::Core::ICoreWindow)->remove_KeyDown(impl::bind_in(cookie)));
    }`

I don't know anything about winrt and have no idea how to adress this problem.
Would anyone help me with this?

dorinung, 2020-11-28:
Can you check that Research mode is enabled in the device portal. See the Setup section in https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ECCV2020-Tutorial/ECCV2020-ResearchMode-Api.pdf (slides 7 and 8)

Thanks

ranchenhao, 2020-11-28:
> Can you check that Research mode is enabled in the device portal. See the Setup section in https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ECCV2020-Tutorial/ECCV2020-ResearchMode-Api.pdf (slides 7 and 8)
> 
> Thanks

Thank you! The problem is solved. I allowed sensor access in "Recording mode" instead of "Research mode".
However, it seems that before I update the system in Hololens, I cannot find the "Research mode", only "Recording mode".

ranchenhao, 2020-11-28:
Thank you! The problem is solved. I allowed sensor access in "Recording mode" instead of "Research mode".
However, it seems that before I update the system in Hololens, I cannot find the "Research mode", only "Recording mode".

Anyway, problem solved. Thank you!

Best regards,
Chenhao Ran

closed 2020-11-28 by ranchenhao