microsoft/HoloLens2ForCV #56: errors while building in VS2019

loleklel, 2020-12-07:
Hi,

I just want to test research mode on my HL2 but in VS2019 I got many errors such as:

Severity	Code	Description	Project	File	Line	Suppression State
Error	C2664	'void BasicHologram::XYZAxisModel::SetColors(DirectX::XMFLOAT3 &,DirectX::XMFLOAT3 &,DirectX::XMFLOAT3 &)': cannot convert argument 1 from 'DirectX::XMFLOAT3' to 'DirectX::XMFLOAT3 &'	CalibrationVisualization	C:\Users\robin.popelka\Desktop\HoloLens2ForCV-main\Samples\CalibrationVisualization\CalibrationVisualization\CalibrationVisualizationScenario.cpp	151	

I tried to build the project within windows SDK 10.0.17* Version and the newest one 10.0.19* but no success.

Can you help me with this issue?

dorinung, 2020-12-11:
I have updated the CalibrationVisualization and SensorVisualization for compile errors that came with the latest version of Visual Studio 16.8. Can you please try the latest changes?

Thanks.

loleklel, 2020-12-11:
Hi, yes it works now with ver. VS 16.8.2

thank you!

dorinung, 2020-12-11:
I have only fixed  CalibrationVisualization and SensorVisualization. I will get to CameraWithCVAndCalibration in a few days.

iimrann412, 2020-12-28:
> I have only fixed CalibrationVisualization and SensorVisualization. I will get to CameraWithCVAndCalibration in a few days.

how did you fix it ? can you tell me please? 
Thanks

raulsf6, 2021-01-08:
I'm getting the same Error C2664 trying to build CameraWithCVAndCalibration in Visual Studio 19

dorinung, 2021-01-11:
I just pushed fixes for CameraWithCVAndCalibration.

Thanks,

raulsf6, 2021-01-12:
Hi @dorinung, It compiled perfectly this time. However, did you try to execute it in Hololens 2? I did it and it just keeps loading. It seems it does not work.

RenSiyao, 2021-02-24:
> 我刚刚推送了CameraWithCVAndCalibration的修复程序。
> 
> 谢谢，

Could you please take a look at this problem for me?thank you
https://github.com/microsoft/HoloLens2ForCV/issues/75#issue-814213398