microsoft/HoloLens2ForCV #61: Where is ResearchModeAPI.dll?

vicobill, 2021-01-04:
When I use `LoadLibraryA("ResearchModeAPI")` in UE4, and add UE4 dll search path "C:/Windows/System32/" manually, or LoadLibrary("C:/Windows/System32/ResearchModeAPI.dll"), it's failed! 

So, what's the location of "ResearchModeAPI.dll", and how can I load this dll in UE4? 

Help me please!

dakesse, 2021-01-04:
In Unity you have to add the dll to the project (_Assets/Plugins/_). Don't you have to do that in UE4 as well?
[Linking DLLs > Unreal Engine Project](https://unreal.gg-labs.com/wiki-archives/devops/linking-dlls#unreal-engine-project)

JiangTao100, 2021-02-05:
It seems that there is only a .h profile named "ResearchModeAPI",  I  wanna import it to Unity,too

dakesse, 2021-02-05:
> It seems that there is only a .h profile named "ResearchModeAPI", I wanna import it to Unity,too

There is no DLL, as this is a C++ project the header is sufficient. If you want to use the researchmode in Unity/C# you will need to build your own DLL based on the [sample code](https://github.com/microsoft/HoloLens2ForCV/tree/main/Samples/SensorVisualization).
There are already some posts regarding this topic: #19, #31, #40

cod3monk3y, 2022-09-27:
@dakesse Each of the four samples in this repo calls `LoadLibraryA("ResearchModeAPI")`, which suggests the area all attempting to load a DLL. But as you say, there is only a header file. See search results: https://github.com/microsoft/HoloLens2ForCV/search?q=LoadLibraryA