microsoft/HoloLens2ForCV #66: OpenCV lib corrupt

abdullahthabit, 2021-01-20:
Hi,

I have been trying to run the CameraWithCVCalibration sample, but I ran into an error with linking the OpenCV static library. It seems the linker doesn't recognize the binaries in the library.

Error: HoloLens2ForCV-main\Samples\CameraWithCVAndCalibration\OpenCvInstallArm64-412d\x64\vc15\staticlib\opencv_world412d.lib : fatal error LNK1107: invalid or corrupt file: cannot read at 0x86

This also happens with the release version of the library: opencv_world412.lib

I am building the solution on a vs 2019. Any idea what is causing the issue? Is it the library itself, or something to do with my linker settings? Any help is appreaciated!

dorinung, 2021-02-03:
Most likely https://git-lfs.github.com/ is needed on your clone.

Thanks

abdullahthabit, 2021-02-07:
It did solve the issue. Thank you @dorinung!

closed 2021-02-07 by abdullahthabit