microsoft/HoloLens2ForCV #70: Updating the building to support research mode

guillermoacdc, 2021-02-08:
Well, i have this build on my device
Windows version19041.**1113.**arm64fre.vb_release_svc_sydney_rel_prod.200808-1020
And the requirements on the tutorial indicate
The earliest build that fully supports research mode is 19041.**1364**

So, I guess that it is necessary to update the windows version on my Hololens2. The alternative to resolve this update is the Windows insider Program. There, I have three options  (Dev, Beta and Release). I have selected the Beta channel and after a restart I have used the windows update. I was expecting to obtain the 19041.1364 version but the update was made for version 19041.**1134**.arm64fre.vb_release_svc_sydney_rel_prod.210111-1541

I suppose that this can be a consequence of 

1. Selection of incorrect channel. So I will change to Dev Channel
2. Language on the OS of my device (currently is spanish). So I will change to english language

Before doing so I would like to have your opinion on this guys

Thanks in advance

JiangTao100, 2021-02-12:
my version is 19041.1136,  but it shows the latest  one and cannot be updated anymore.

guillermoacdc, 2021-02-12:
"Research-Mode". See attached image

**My interpretation**: The OS of my device is not updated, so the "research-mode" option is missing

Currently my device has buildings of 19041.1113..., and the tutorial indicates that "earliest build that fully supports research mode is 19041.1364"

**What I have done**:  update the OS version using Windows insider Program / Beta Channel.

After this I have advanced to building 19041.1134..., not enough to the reference in the tutorial. The "Research-Mode" is still missing in my device portal

**My real question**: How can I update my OS to building 19041.1364 or later

**My current OS version**: 19041.1134, Spanish language

[image]

guillermoacdc, 2021-02-14:
Hello guys

I have updated my language settings to English and now the "Research Mode" is available in the Device-Portal. Thks!
Build

19041.1134.arm64fre.vb_release_svc_sydney_rel_prod.210111-1541
[image]

closed 2021-09-10 by guillermoacdc