microsoft/HoloLens2ForCV #71: Sensor device consent issue

NHarishGit, 2021-02-11:
**Overarching goal**: Get IMU data from the modified HL2 device.

Trying to run the latest main trunk code on my modified HL2 device but it keeps failing at this line:

`winrt::check_hresult(m_pSensorDevice->QueryInterface(IID_PPV_ARGS(&m_pSensorDeviceConsent)));`

**My interpretation**: Since my HL2 has different sensors the "device consent" piece is NOT working as it may be asking consent using GUIDs which are NOT applicable in my hardware. it is just my guess. 

**My real question**: How does one get GUIDs such as below "EAB9D672-9A88-4E43-8A69-9BA8f23A4C76" ?:

`DECLARE_INTERFACE_IID_(IResearchModeSensorDeviceConsent, IUnknown, "EAB9D672-9A88-4E43-8A69-9BA8f23A4C76")
{
    STDMETHOD_(HRESULT, RequestCamAccessAsync)(void (*camCallback)(ResearchModeSensorConsent))= 0;
    STDMETHOD_(HRESULT, RequestIMUAccessAsync)(void (*imuCallback)(ResearchModeSensorConsent)) = 0;
};`

dakesse, 2021-02-11:
Research Mode - and thus IMU access - is only supported with **19041.1364** or newer. Have a look in the [doc file](https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ResearchMode-ApiDoc.pdf) or issues #70.

guillermoacdc, 2021-02-12:
**Overarching goal**: Enable Research Mode in my Hololens2.

When following instructions in the tutorial at xxx i have found that my "Device-Portal" doesnt include the option "Research-Mode". See attached image

**My interpretatio**n: The OS of my device is not updated, so the "research-mode" option is missing

Currently my device has buildings of 19041.1113..., and the tutorial indicates that "earliest build that fully supports research mode is 19041.1364"

**What I have done**:  update the OS version using Windows insider Program / Beta Channel.

After this I have advanced to building 19041.1134..., not enough to the reference in the tutorial. The "Research-Mode" is still missing in my device portal

**My real question**: How can I update my OS to building 19041.1364 or later

**My current OS version**: 19041.1134, Spanish language

dakesse, 2021-02-15:
> Research Mode - and thus IMU access - is only supported with **19041.1364** or newer. Have a look in the [doc file](https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ResearchMode-ApiDoc.pdf) or issues #70.

But as far as I remember research mode was already available from Windows Holographic, version 2004 - July 2020 Update (19041.1109). However it was only communicated from Windows Holographic, version 20H2 (19041.1128) in the [release notes](https://docs.microsoft.com/en-us/hololens/hololens-release-notes).