microsoft/HoloLens2ForCV #75: CalibrationVisualization  Project Release Issues

RenSiyao, 2021-02-23:
When building the project 'Samples\CalibrationVisualization, the VS2019 release code tells you that the initial value of the non-trivial reference must be an lvalue.Others cannot find identifiers and others do not declare identifiers.This caused me to be unable to publish successfully, do I need to set anything else?
[image]
[image]

RenSiyao, 2021-02-24:
There is also a hint that this method needs a return value, I don't know what to do
[image]
[image]

dorinung, 2021-02-24:
Do you have the latest changes in the repo? I believe these where fixed with:
https://github.com/microsoft/HoloLens2ForCV/commit/c316733afcd059d6b6f460e8c186cddda6fe185b
Thanks.

RenSiyao, 2021-02-25:
> 您是否在回购中有最新更改？我相信这些固定在：
> [c316733](https://github.com/microsoft/HoloLens2ForCV/commit/c316733afcd059d6b6f460e8c186cddda6fe185b)
> 谢谢。

I already have the latest changes, but still you will be prompted me when released in "CalibrationVisualizationScenario. CPP" "STD: : future&lt;void&gt; ModelRenderer: : CreateDeviceDependentResources () method must return a value.