microsoft/HoloLens2ForCV #76: Can't access data captured by StreamRecorder

sitzikbs, 2021-03-04:
HI,
I am testing out the StreamRecorder app. Was able to run fine on the device. Trying to get the data off the device and it doesn't work. 
a. Using the device portal I get "failed to get files for package: ..." 
b. Using the python code - what is the path? 

I tried connecting via wifi and USB (the appdata folder is empty in USB but in wifi device portal it is there)

How can I get the data?

ylsislove, 2021-03-04:
@sitzikbs  Hi, maybe you can look at the `StreamRecorder\README.md`, which provides a lot of helpful information. Actually, you can use `python StreamRecorderConverter/recorder_console.py --workspace_path <output_folder>  --dev_portal_username <user> --dev_portal_password <password>` to download data from hololens, and this #4 maybe helpful.

sitzikbs, 2021-03-04:
Thanks!
 #4 worked (uncheck SSL connection from Preferences under system at Windows Device Portal). 
Also, I found the documentation for `workspace_path` confusing - I thought it meant the path on the device (not on the local machine - but maybe it is just me). 

closing now.

closed 2021-03-04 by sitzikbs