microsoft/HoloLens2ForCV #79: IMU lags behind Camera

kellenberger, 2021-03-15:
I have an application similar to StreamRecorder where I access the camera and some IMU sensors (Accelerometer and Gyroscope). The application works in realtime and as suggested in #28 I retrieve all sensor data in seperate threads and unite them in an additional thread.

However the IMU data lags behind the camera frames by up to 0.3s (when obtaining host ticks in the same way as in the StreamRecorder app). This of course is way to long for a realtime application. Is there a way to reduce this lag? I have checked and the processing done by the application is not the issue, the lag is already present once the data arrives. Any help would be appreciated.

Sayoj, 2021-04-26:
Hey @kellenberger 

I am actually working on a similar code, but having trouble with transmitting the IMU sensors in real time. Would you be so kind and share your code with us? So we can maybe help each other.

Kind regards

kellenberger, 2021-05-13:
Hi @Sayoj, unfortunately I am not allowed to share my code and am currently no longer working on this. However, as far as I know the problem still persists.