microsoft/HoloLens2ForCV #86: Stream Recorder: VLC frames have incorrect timestamps (frames out of order)

nickgreene, 2021-05-17:
When recording using StreamRecorder it seems that the timestamps returned by all of the visible light cameras are frequently incorrect resulting in an out of order sequence of frames when sorting by timestamp.

To confirm this, I made a recording while looking at a stopwatch with each visible light camera. There are many frames where the reading on the stopwatch is lower than the previous frame (despite the frame's larger timestamp). Sometimes a disruption lasting more than 5 frames happens before the timesteps seem to become correct again.

This happens often enough for it to be a serious inconvenience. For a rough idea, I counted 12 times where the stopwatch time decreased compared to the previous frame in a 10 second recording. The resulting video playback has lots of noticeable jitters.

I tried using ```timestamp.SensorTicks```  instead of ```timestamp.HostTicks``` in ```RMCameraReader.cpp``` but the issue persisted.

This does not happen for the PV frames or for either mode of the depth sensor frames.

I'm using the latest insider preview build: Windows Version 21H1, OS build 20346.1402

I may be wrong but I do not recall this issue occurring with the first few insider builds that supported research mode, however I couldn't find the older builds online to try.

Is there any way to fix this issue?

Thanks a lot!

evia100, 2021-12-07:
same problem here.
did you managed to make it work?

jondebug, 2021-12-08:
same here..
any update??