microsoft/HoloLens2ForCV #91: opencv release version for  CameraWithCVAndCalibration

bo-wu, 2021-06-02:
Hi authors,

Thanks a lot for the great repository. I have successfully built the CameraWithCVAndCalibration sample in debug version. However, it seems the opencv_world412.lib for release version is missing.  Where can I get the library, or do I have to compile it by myself ?

lihaowei1999, 2021-07-18:
It seems that the lib is in directory **CameraWithCVAndCalibration\OpenCvInstallArm64-412d\x64\vc15\lib** the problem you said can be solved by adding this path to the environment in your project, and can deploy successfully after this alteration. But the it seems that the program lack some dll after deploy.

bo-wu, 2021-08-19:
@lihaowei1999 
Thanks for the reply. However, it is the DEBUG version of opencv, not RELEASE version.

bo-wu, 2021-08-23:
Close. See Issue #96

closed 2021-08-23 by bo-wu