microsoft/HoloLens2ForCV #96: CameraWithCVAndCalibration not working!

Julian243, 2021-07-09:
Hello,

I've been trying to get the CameraWithCVAndCalibration sample to work for the past two days. However I still haven't succeded yet.
Opening the sln-file I got the error: LNK1104 cannot open file 'opencv_world412d.lib' 	so I manually added the path to the .lib-file under Properties->Configuration Properties->VC++ Directories->Library Directories.
Trying to deploy the project to HoloLens2 as Debug|ARM64 configuration results in the following two errors:
[image]
[image]
And yes I did use git lfs clone to clone the repository!
Can anyone please help me? I spend hours trying to figure out where things went wrong and still have no clue... I tried out different PC's and even factory resetted my HoloLens2. Other examples work just fine but I can't get this to work.

Thanks in advance!

Julian243, 2021-07-12:
Edit: The console shows the following output. Could it be that there might be a dll reference missing?
[image]

marcfischer123467, 2021-07-14:
I have the same issue

lihaowei1999, 2021-07-21:
I have the same issue too.

Julian243, 2021-07-23:
Update: Checking out an earlier version of this repository solved the issue for me! I checked out commit 41ec9e542a7a84b3a68e5089b9906864bc7c526a which contains the missing lib files in the staticlib folder.
Please fix this!

lihaowei1999, 2021-07-23:
> Update: Checking out an earlier version of this repository solved the issue for me! I checked out commit [41ec9e5](https://github.com/microsoft/HoloLens2ForCV/commit/41ec9e542a7a84b3a68e5089b9906864bc7c526a) which contains the missing lib files in the staticlib folder.
> Please fix this!

I'vs also solved this problems through this method !!!!

bo-wu, 2021-08-23:
Thanks a lot for the solution. Don‘t understand why the authors delete the library.

dogadogan, 2021-12-02:
Downloading the previous commit didn't help with the OpenCV lib issue. :/ Could anyone clarify how they got the CameraWithCVAndCalibration app to work?
Thank you in advance!

shoukailiang, 2021-12-30:
> Downloading the previous commit didn't help with the OpenCV lib issue. :/ Could anyone clarify how they got the CameraWithCVAndCalibration app to work? Thank you in advance!

Have you solved it? I have the same problem?

shoukailiang, 2022-01-01:
the commit `41ec9e5 ` did work!!!

friederwC, 2022-04-12:
Thanks for the pointer for using this older commit. Hope the fixing PR https://github.com/microsoft/HoloLens2ForCV/pull/120 gets merged soon. 

I'm able to build the app now, but get the following run time error after deploying to my HL2:

```
Exception thrown at 0x00007FFD9D8C5E5C (KernelBase.dll) in CameraWithCVAndCalibration.exe: WinRT originate error - 0x8000FFFF : 'Catastrophic failure'.
Exception thrown at 0x00007FFD9D8C5E5C in CameraWithCVAndCalibration.exe: Microsoft C++ exception: winrt::hresult_error at memory location 0x00000021234FD290.
```

Anyone else has this problem.

frank-Regal, 2023-01-20:
@friederwC I had the same problem. I realized research mode was not enabled on my HoloLens 2. Make sure you have research mode enabled.