microsoft/HoloLens2ForCV #99: Temperatur sensor/ accuracy depth sensor

marcfischer123467, 2021-07-25:
Hi, 
Thank you for this amazing repo !
I was wondering if someone in the community might know if there is any way to get a temperature reading from the HoloLens 2 to measure if temperature affects the accuracy of the depth sensor? Are there any internal temperature sensors one could access that would be representative of the temperature of the depth sensor?

Thanks!

kianwei96, 2021-07-26:
> Hi,
> Thank you for this amazing repo !
> I was wondering if someone in the community might know if there is any way to get a temperature reading from the HoloLens 2 to measure if temperature affects [...]

Hi @marcfischer123467 , you can access the temperature data of the gyroscope/accelerometer as a proxy reading. Those can be accessed through research mode. I've noticed a significant depth sensor bias sometimes, so very interested in whether you can find any trend :)

you might find this useful - https://github.com/petergu684/HoloLens2-ResearchMode-Unity/issues/13

kysucix, 2021-07-29:
There's no way to measure the temperature of the depth sensor. Probably the best way is with IMU, as kianwei64 suggested or otherwise you might check standard windows api to read temperature sensors.

closed 2021-07-29 by kysucix