microsoft/HoloLens2ForCV #108: How to rectify depth images from ATAH/Long Throw camera sensors of HoloLens2?

soberkiki, 2021-11-14:
Hello.When I reconstructed the point cloud via the sensor of the depth camera, the point cloud was inaccurate due to the distortion of the camera.Does anyone have any methods or suggestions on how to rectify images taken by depth cameras.The images below are point cloud views from three perspectives.
[image]

[image]
[image]

closed 2021-11-15 by soberkiki

agroenenberg, 2022-04-28:
@soberkiki I seem to be having the same problem when reconstruction without the lookuptable, did you find a solution?

felixvh, 2022-04-29:
The lookup table is supposed to be used for this reason. Why do you want to reconstruct it without the LUT?

agroenenberg, 2022-04-29:
I need to do some recalculations for my research, but I found it! Simple mistake, thanks!