microsoft/HoloLens2ForCV #111: Recorder Console - GetAddrFailed

lpug21, 2021-11-22:
Hey all.  Using the following command:  

`python recorder_console.py --dev_portal_address 192.168.10.64 --dev_portal_username dandandan --dev_portal_password dandandan --workspace_path output`

Returns error 
`urllib.error.URLError: <urlopen error [Errno 11001] getaddrinfo failed>`

Other listed usage does not specify using the dev portal address [(as per the documentation here)](https://github.com/microsoft/HoloLens2ForCV/tree/main/Samples/StreamRecorder) suggests using the below commands:
`python StreamRecorderConverter/recorder_console.py --workspace_path <output_folder>  --dev_portal_username <user> --dev_portal_password <password>`

However, this returns an issue - target machine actively refused the request.  Any insights here?

lpug21, 2021-11-22:
Resolved this myself:  

In recorder_console.py I hardcoded the URL to the device portal URL (In my case, "https://192.168.10.64".  I then ran the second usage scenario (path, output, username, password) - and it worked well!

closed 2021-11-22 by lpug21

GLARKI, 2022-11-16:
You mean the self.url right? and then "https://ip".format(address)
Now the error changes to 
urllib.error.URLError: <urlopen error [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond>

    parser.add_argument("--dev_portal_address", default="127.0.0.1:10080",
                        help="The IP address for the HoloLens Device Portal")
                        
        self.url = "https://10.35.137.104" #.format(address) # "http://{}".format(address)

GLARKI, 2022-11-16:
That (inserting the IP) solved it partially. Thereafter I got an SSL Certification error. 
But if you just connect it (HL2) via usb-c, it works like a charm. I no this is a workaround, but I could not fix the certification error and happy it's working. For HL2 and recorder_console, this should suffice.