microsoft/HoloLens2ForCV #118: Can not connect and download via recorder_console.py in StreamRecorderConverter

friederwC, 2021-12-09:
I can access my Holoelens 2 via WiFi and the Windows Device Portal in the browser. This way I can download the recordings of StreamRecorder. However, the provided script for automatic data download does not work.

This is the command I use, with the ip address (172.16.11.158), username and password which works with Windows Device Portal via the browser. For the port number (9000) I tried also with 10080

`python recorder_console.py --dev_portal_address '172.16.11.158:9000' --dev_portal_username 'username' --dev_portal_password 'password' --workspace_path 'C:\example'
`
The error message is:

`Connecting to HoloLens Device Portal...
[...]
TimeoutError: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond
[...]
urllib.error.URLError: <urlopen error [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond>
`
I also tested with a turned of requirement for SSL (Device Portal - > System -> Preferences) but I get the same error message.

evia100, 2022-01-02:
Try without the port number, It worked for me.

friederwC, 2022-04-12:
Removing the port number worked.

closed 2022-04-12 by friederwC