microsoft/HoloLens2ForCV #122: stream recoder UI and data sjze

km0703, 2021-12-20:


evia100, 2022-01-04:
Hi, did you find any solution?

km0703, 2022-01-04:
Hi, no I did not.

friederwC, 2022-04-12:
Hi Minato

You can try the patch below for removing the buttons and automatic recording of a fixed number of frames at the start of the app. You can conviniently start and close the app via the Device Portal. It's hacky but does the job.

I did not test for longer recordins though, so I can't help you with your second issue.

```
Subject: [PATCH] automatic recording at start of app for 300 frames

---
 .../StreamRecorderApp/AppMain.cpp             | 38 ++++++++++++++++---
 1 file changed, 32 insertions(+), 6 deletions(-)

diff --git a/Samples/StreamRecorder/StreamRecorderApp/AppMain.cpp b/Samples/StreamRecorder/StreamRecorderApp/AppMain.cpp
index b0effed..c041ed3 100644
--- a/Samples/StreamRecorder/StreamRecorderApp/AppMain.cpp
+++ b/Samples/StreamRecorder/StreamRecorderApp/AppMain.cpp
@@ -42,6 +42,9 @@ std::vector<ResearchModeSensorType> AppMain::kEnabledRMStreamTypes = { ResearchM
 }*/
 std::vector<StreamTypes> AppMain::kEnabledStreamTypes = { StreamTypes::PV };
 
+int const NUMBER_OF_FRAMES_TO_RECORD = 300;
+int numberOfFramesRecorded = 0;
+
 AppMain::AppMain() :
 	m_recording(false),
 	m_currentHeight(1.0f),
@@ -67,8 +70,8 @@ AppMain::AppMain() :
 	m_menu.HideTitleBar();
 	m_menu.SetSize(XMVectorSet(0.15f, rootMenuHeight, 0.01f, 1.0f));
 
-	m_menu.AddButton(make_shared<FloatingSlateButton>(XMVectorSet(-0.025f, 0.0f, 0.0f, 1.0f), mainButtonSize, XMVectorSet(0.0f, 0.5f, 0.0f, 1.0f), (unsigned)ButtonID::Start, this, "Start"));
-	m_menu.AddButton(make_shared<FloatingSlateButton>(XMVectorSet(0.025f, 0.0f, 0.0f, 1.0f), mainButtonSize, XMVectorSet(0.5f, 0.0f, 0.0f, 1.0f), (unsigned)ButtonID::Stop, this, "Stop"));
+	//m_menu.AddButton(make_shared<FloatingSlateButton>(XMVectorSet(-0.025f, 0.0f, 0.0f, 1.0f), mainButtonSize, XMVectorSet(0.0f, 0.5f, 0.0f, 1.0f), (unsigned)ButtonID::Start, this, "Start"));
+	//m_menu.AddButton(make_shared<FloatingSlateButton>(XMVectorSet(0.025f, 0.0f, 0.0f, 1.0f), mainButtonSize, XMVectorSet(0.5f, 0.0f, 0.0f, 1.0f), (unsigned)ButtonID::Stop, this, "Stop"));
 
 	m_hethateyeStream.Clear();
 
@@ -102,10 +105,33 @@ void AppMain::Update()
 	m_mixedReality.Update();
 	m_hands.UpdateFromMixedReality(m_mixedReality);
 
-	auto startButton = m_menu.GetButton((unsigned)ButtonID::Start);
-	auto stopButton = m_menu.GetButton((unsigned)ButtonID::Stop);
-	startButton->SetDisabled(m_recording || (!IsVideoFrameProcessorWantedAndReady()));
-	stopButton->SetDisabled(!m_recording);
+	if (numberOfFramesRecorded == 0 && IsVideoFrameProcessorWantedAndReady()) 
+	{
+		m_hethateyeStream.Clear();
+		m_hethatStreamVis.Update(m_hethateyeStream);
+		SetDateTimePath();
+		StartRecordingAsync();
+
+		m_recording = true;
+	}
+
+	if (m_recording && numberOfFramesRecorded < NUMBER_OF_FRAMES_TO_RECORD)
+	{
+		numberOfFramesRecorded++;
+	}
+	else 
+	{
+		if (m_recording) 
+		{
+			StopRecording();
+		} 
+
+		m_recording = false;
+	}
 
 	const XMVECTOR headPosition = m_mixedReality.GetHeadPosition();
 	const XMVECTOR headForward = m_mixedReality.GetHeadForwardDirection();
-- 
2.33.0.windows.2

```

km0703, 2022-05-26:
Hi friederwC 
I appreciate your help.
We try this code and finally we can delete buttons!
Please let me express my sincere gratitude once again.

kjayarajah, 2023-07-02:
Has anyone had any luck trying to do long recordings? I'm unable to record anything longer than 1-2 minutes!