microsoft/HoloLens2ForCV #136: StreamRecorder Recorded Data

sgessman, 2022-06-09:
Hi,
I am having a problem with accessing the recorded data.  Following the instructions on the repository the captures should be showing up in device portal by going to System-->FileExplorere --> LocalAppData --> StreamRecorder -->LocalState.  The folder shows up but when I go to click on them I get the following error that failed to get files for package.  Does anyone know how to actually be able to view the captured streams? 

Thank you for the help!

ArmandB, 2022-06-12:
I'd recommend using the "StreamRecorderConverter". There's a python script "recorder_console.py" that's really useful for pulling the data from the HL2. 

I think you have to change the IP address in the script and lower some security barriers in the device portal, but even if you could see the files in the fileExplorer, manually downloading big files through the device portal usually hangs.

Then, once they're on your computer: tar -xvf <file.tar> will get you the images.

sgessman, 2022-06-13:
Thank you so much! I'm new to using the HoloLens and visual studios, how do I deploy the script as well as tar?

sgessman, 2022-06-15:
Thank you so much! Where in the script do I change the IP address?

closed 2022-06-17 by sgessman