microsoft/HoloLens2ForCV #142: Access Violation when Accessing Depth Long Throw Sensor and PV

jonashein, 2022-07-18:
Hi all,

I'm randomly getting the following Access Violation Error when trying to get the next frame from the Depth Long Throw sensor:
`Exception thrown at 0x00007FFDF771C480 (ResearchModeAPI.dll) in StreamRecorder.exe: 0xC0000005: Access violation reading location 0x0000000000000000.`
The exception is thrown in [line 85 of the CameraUpdateThread function](https://github.com/microsoft/HoloLens2ForCV/blob/207205596840ae6e7b8c2a795d35c3c4e7bec22e/Samples/StreamRecorder/StreamRecorderApp/RMCameraReader.cpp#L85).  Interestingly, the exception is not thrown instantly: Sometimes the app runs fine for >30 minutes, sometimes the exception occurs within the first 5 or 10 minutes of a recording. This bug seems to be related to #133, though I'm not accessing the AHAT sensor but the Depth Long Throw. 

I'm running a slightly customized version of the StreamRecorderApp, without any relevant changes to the RMCameraReader. Besides the Depth Long Throw sensor, I'm also capturing PV camera, hands&eye gaze, as well as the front-facing visible light cameras. 
The exception does not seem to occur when I don't capture PV and Depth Long Throw simultaneously, though I cannot be certain due to the inconsistent nature of this bug. According to the Windows Device Portal, there's always at least 1GB of RAM available, usually ~1.5GB.

I can reproduce the exception on two different devices, as well as under HoloLens OS version 20348.1432.arm64fre.fe_release_svc_sydney_rel_prod.211005-1308 and 20348.1506.arm64fre.fe_release_svc_sydney_rel_prod.220607-1410. 

Any help will be appreciated!

Best,
Jonas

mahdidBagheri, 2023-08-16:
Exactly!!! To be honost, I am working with Hololens2 for a bout 1.5 years and one the most astonishing issues is this one. Among 3 devices in which I've worked with, **ALL 3** had the same issue in a shocking way! Reading PV and Research Mode Sensors encountered no problem for couple of early weeks that I started to work with. Suddenly, at an unknown time they start to fail on giving PV and Research Mode Sensors simultaneously! I've even tried to synchronize the threads when they request the data from their APIs but none of that worked! It seems conflict roots when they initialize! I've even not been able to watch the stream through Windows Portal!

jdibenes, 2023-08-18:
I've experienced the same problem. The issue seems to go away if the research mode streams are opened and the first frame is acquired before opening the next stream. Like open front left grayscale stream and acquire first front left frame, then open depth longthrow and acquire first frame, and so on, then finally create the PV frame reader and start capturing PV frames.