microsoft/HoloLens2ForCV #144: StreamRecorder Start and Stop

sgessman, 2022-08-09:
Hello, I was wondering if anyone knew of a way to have the Stream Recorder app start and stop another way not using the holographic start and stop buttons.  I am using the HoloLens to record some experiments.  Currently, I have the device set up on a mount to take the recordings but I have to first put them on myself to start the recording and then place them on the mount which isn't ideal.  Thank you in advance!

guillermoacdc, 2022-09-05:
I have a similar problem, I am needing to synchronize the start of acquisition of the devices (1) Hololens 2 - stream recorder app and (2) kinect azure. I know the azure kinect has a port to facilitate this kind of [sync]([image]) . Do you know of any way to start the acquisition synchronously from hololens and with the TTL signal of the azure kinect?

guillermoacdc, 2022-09-15:
There is an alternative here--->
#122