microsoft/HoloLens2ForCV #148: How to get depth camera data in unity realtime?

Hans86, 2022-09-20:
As title, how to get depth camera data using unity with openxr to build a holographic application?
Is there any method to get the data stream?
I have tried this method(https://github.com/cgsaxner/HoloLens2-Unity-ResearchModeStreamer) but if I put it into my project , then it will crash soon and I never get depth camera data successfully...

revkusz, 2022-10-11:
Currently there is no official way (not that i know of) to get the ResearchModeApi data directly to c# or unity. If those are not working out of the box then they need a little tinkering as always with some of the Hololens 2 sample applications. I could never make those work at first try for some reason.

vbandi, 2022-10-13:
I have a repo that does this (based on another PeterGu's repo). Maybe you can get that one to work? 
https://github.com/vbandi/HL2-ResearchMode