microsoft/HoloLens2ForCV #153: CameraWithCVAndCalibration runtime error: Unable to activate Microsoft Store app

nothingbut1, 2022-12-07:
Hello and thanks for these useful samples. 

**I am able to run all the samples except "CameraWithCVAndCalibration"**. 

Here is my procedure:
1. Open the CameraWithCVAndCalibration.sln file in Visual Studio, build (Debug, ARM64), errors occured：
[image]
so I check the property settings, and add the opencv_world412d.lib path to the Library Directories:
[image]
rebuild, succedded with two warnings (these warning matters ?):
[image]

2. Deploy succeeded.
[image]

3.  Press F5 to debug, an error window displayed as follows:
[image]
press OK, another error window displayed as follows:
[image]
 and in the visual studio output window shows:
[image]
**It seems that there are some errors with dll library**, but I don't know how to fix it.

 I have refered to the issues #8 and #41 with similar error, but it doen't works for me. 
 Device Portal and Research Mode are all enabled.
 [image]

Environment：
Hololens: beta developer mode, Windows version 22621.1036.arm64fre.ni_release_svc_sydney.221129-1357
visual studio 2022 professional

Any help would be grately appreciated!

joriswa, 2022-12-07:
The opencv lib is stored with Git LFS so you need clone it with Git LFS. Otherwise the file will be empty

nothingbut1, 2022-12-12:
see #96

closed 2022-12-12 by nothingbut1